dotnet/iot #2134: Mcp23017 missing Events/RegisterCallbackForPinValueChangedEvent

wolfgangmauer, 2023-09-20:
protected override void AddCallbackForPinValueChangedEvent(int pinNumber, PinEventTypes eventTypes,
    PinChangeEventHandler callback) => throw new NotImplementedException();

/// <inheritdoc/>
protected override void RemoveCallbackForPinValueChangedEvent(int pinNumber, PinChangeEventHandler callback) =>
    throw new NotImplementedException();

/// <inheritdoc/>
protected override WaitForEventResult WaitForEvent(int pinNumber, PinEventTypes eventTypes,
    CancellationToken cancellationToken) => throw new NotImplementedException();

pgrawehr, 2023-09-26:
You are right, and this should probably be improved or at least better documented. There's an easy workaround, though. I'll need to look up the correct example code.

pgrawehr, 2023-09-27:
So, now I've got it present. To use interrupt handling with an Mcp23017, you need to first make sure you have one or both interrupt lines connected to the master GPIO controller (on a Raspberry Pi, connect the InterruptA or InterruptB output to a free GPIO pin). Then you can do something like the following:

```
_gpioController = new GpioController();
_device = I2cDevice.Create(new I2cConnectionSettings(1, 0x21));
// Interrupt pin B is connected to GPIO pin 22
_mcp23017 = new Mcp23017(_device, -1, -1, 22, gpioController, false);
_mcp23017.EnableInterruptOnChange(8, PinEventTypes.Rising | PinEventTypes.Falling); // Enable interrupt for pin 8
_gpioController.RegisterCallbackForPinValueChangedEvent(22, PinEventTypes.Falling, Interrupt);
// Read the interrupt register, to make sure we get any further interrupts
_mcp23017.ReadByte(Register.GPIO, Port.PortB);
```

with the `Interrupt` function defined as follows:
```
private void Interrupt(object sender, PinValueChangedEventArgs pinvaluechangedeventargs)
        {
            // We don't really care what caused the interrupt to trigger - we just read if we receive any interrupt.
            lock (_counterLock)
            {
                int newValue = _mcp23017.ReadByte(Register.GPIO, Port.PortB); // Make sure to read this register on every interrupt, as this resets the interrupt flag (otherwise, the interrupt would trigger just once)
                // ....
            }
        }
```

(I've copied this from another larger piece of code, so it might not be perfectly accurate)

wolfgangmauer, 2023-09-27:
THX,
but where this "EnableInterruptOnChange" come from?
I use 
```
<PackageReference Include="System.Device.Gpio" Version="3.0.0" />
<PackageReference Include="Iot.Device.Bindings" Version="3.0.0" />
```
and there is no "EnableInterruptOnChange" :-(

pgrawehr, 2023-10-01:
@wolfgangmauer I'm sorry, I wasn't aware that this method only exists on my fork (as I had written it long ago). I'll create a PR to add it immediately.

wolfgangmauer, 2023-10-07:
@pgrawehr THX a lot !!!

I use this board 
[https://www.waveshare.com/wiki/MCP23017_IO_Expansion_Board](https://www.waveshare.com/wiki/MCP23017_IO_Expansion_Board)

```
mcp23017 = new Mcp23017(I2cDevice.Create(new I2cConnectionSettings(1, 0x27)), -1, 4, -1, gpioController, false);
var theDeviceController = new GpioController(PinNumberingScheme.Logical, mcp23017);
theDeviceController.OpenPin(0, PinMode.Input);
theDeviceController.RegisterCallbackForPinValueChangedEvent(0, PinEventTypes.Rising, Callback);

private static void Callback(object sender, PinValueChangedEventArgs pinValueChangedEventArgs)
{
    Console.WriteLine($"{DateTime.Now} PinNumber:{pinValueChangedEventArgs.PinNumber} ChangeType:{pinValueChangedEventArgs.ChangeType}");
}
```

Maybe I'm wrong, but the mcp23017 has pullup/pulldown support (?)
System.InvalidOperationException: Pin 1 does not support mode InputPullUp.

Another problem is, when the pin is not connected i get ongoing Rising/Falling, even if the PinEventTypes.Rising only is set :-(

```
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Falling
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Falling
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Rising
07.10.2023 19:54:47 PinNumber:0 ChangeType:Falling
```

wolfgangmauer, 2023-10-07:
@pgrawehr IMHO it is necessary to use a pullup for inputs, this also applies to the input port of the pi. A normal input makes no sense at all because the state changes uncontrollably. The pullup keeps the input at VCC with a resistor and a button then closes the circuit to ground. Works on the PI(3/4) with InputPullUp perfect.

pgrawehr, 2023-10-08:
@wolfgangmauer The first error (wrong ChangeType) is due to the way this is detected. Since there's no way of detecting whether the interrupt was caused by a falling or a rising edge, I'm just reading out the current state of the pin and decide that it must have been a rise if the value is now 1 (and vice-versa). If the pin flips very quickly, this might be wrong. If only one edge is enabled, this could be prevented, but increases complexity of the implementation significantly. 

In my opinion, it's easier to just assume that a falling event must have been preceeded by a rising one, so if you have requested only one type, there must have been (at least one) event of the given type. Note also that there's a timing limit: Very fast changing inputs will not result in every change triggering an event, because the interrupt flag must have been read before a new one can be triggered. If you have very fast changing inputs you need to count, you need a counter IC (instead, or in addition). 

You are right, the MCP supports internal pull-ups, but this feature was just not implemented. Should be easy to add. Note that not all sensors need that, some have it built-in and then an additional pull-up might create issues. Also, you can always use an external resistor instead of the internal pullup.

wolfgangmauer, 2023-10-08:
@pgrawehr "Should be easy to add." -> Ok, i wait ;-) .....

pgrawehr, 2023-10-08:
Done. Please test.

wolfgangmauer, 2023-10-08:
@pgrawehr Where? PR ?

wolfgangmauer, 2023-10-08:
@pgrawehr If the Input is open(not connected) with InputPullUp :-(

08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling

wolfgangmauer, 2023-10-08:
@pgrawehr Register.GPPU doesn't work at all, with Register.GPPUA(0x0c) and Input is open(not connected) with InputPullUp :-(

08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling
08.10.2023 20:52:16 PinNumber:0 ChangeType:Rising
08.10.2023 20:52:16 PinNumber:0 ChangeType:Falling

wolfgangmauer, 2023-10-08:
@pgrawehr Seems that 8-Bit and 16-Bit Boards needs diffrent implementation (enum Register, etc.)

wolfgangmauer, 2023-10-09:
@pgrawehr THX for your work !!!!

Test:
Direct connected
PA0 -> PB0
PA1 -> PB1

```
Console.WriteLine("Hello, World!");
var mcp23017 = new Mcp23017(I2cDevice.Create(new I2cConnectionSettings(1, 0x27)), -1, 4, -1);
var theDeviceController = new GpioController(PinNumberingScheme.Logical, mcp23017);

theDeviceController.OpenPin(8, PinMode.Output);
theDeviceController.OpenPin(9, PinMode.Output);

theDeviceController.OpenPin(0, PinMode.InputPullUp);
theDeviceController.RegisterCallbackForPinValueChangedEvent(0, PinEventTypes.Rising | PinEventTypes.Falling, Interrupt);
theDeviceController.OpenPin(1, PinMode.InputPullUp);
theDeviceController.RegisterCallbackForPinValueChangedEvent(1, PinEventTypes.Rising | PinEventTypes.Falling, Interrupt);

Task.Factory.StartNew(DoWorkAsync);

await Task.Delay(Timeout.Infinite);

void Interrupt(object sender, PinValueChangedEventArgs pinValueChangedEventArgs)
{
    Console.WriteLine("Interrupt");
}
async Task DoWorkAsync()
{
    while (true)
    {
        Console.WriteLine("Write low");
        theDeviceController.Write(8, PinValue.Low);
        theDeviceController.Write(9, PinValue.Low);
        await Task.Delay(1000);
        Console.WriteLine("Write high");
        theDeviceController.Write(8, PinValue.High);
        theDeviceController.Write(9, PinValue.High);
        await Task.Delay(1000);
    }
}
```

the output:

```
Hello, World!
Write low
Write high
Interrupt
Write low
Interrupt
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low
Write high
Write low

```

pgrawehr, 2023-10-09:
Ah, I believe this could be because the class is not thread safe, which includes the callback method itself.

pgrawehr, 2023-10-10:
I will certainly give it a try. Shouldn't be too difficult, except that I need to care for  deadlocks.

wolfgangmauer, 2023-10-11:
@pgrawehr If the Input trigger come to fast(e.g. PWM) maybe a queue for the "CallHandlerOnPin" is the goal

pgrawehr, 2023-10-11:
A queue would help to make sure we get all the trigger (or most, after all), but your issue seems to be that you get no triggers at all when it's fast. This shouldn't happen and is presumably some race condition.

pgrawehr, 2023-10-13:
@wolfgangmauer I'm a bit confused: Did you delete some comment about the pull-up feature? Is that part working now? I checked the code again and cannot see why it wouldn't. 

I'll have a look at the locking issue till sunday.

wolfgangmauer, 2023-10-14:
@pgrawehr Yes, i deleted some "wrong" comments.
I tested the pull-up feature with your PR and it words perfect!

wolfgangmauer, 2023-10-15:
@pgrawehr Works better, but not quite yet
```
Hello, World!
Write low
Write high
Interrupt: Rising PIN: 0
Write low
Interrupt: Falling PIN: 1
Write high
Interrupt: Rising PIN: 0
Write low
Interrupt: Falling PIN: 1
Write high
Interrupt: Rising PIN: 1
Write low
Interrupt: Falling PIN: 1
Write high
Interrupt: Rising PIN: 1
Write low
Interrupt: Falling PIN: 1
Write high
Interrupt: Rising PIN: 1
```

pgrawehr, 2023-10-15:
Hmm... That is a limitation of the device. If an interrupt occurs while another interrupt is still pending (and has not been handled), the second interrupt is lost. Since your example writes pins 8 and 9 almost simultaneously, this will only trigger one interrupt. Could be a bit tricky to work around this.

wolfgangmauer, 2023-10-15:
@pgrawehr I think that all Read/Write must be "blocked" until "InterruptHandler" is finish.
The "while (mask < 0x10)" invoke all events within one port at once, right?

pgrawehr, 2023-10-15:
Yes, but after reading the documentation again now, it seems that only one port will report an event until the interrupt is cleared. The INTF register seems to only ever have a single bit set, so I cannot use this as indication to handle many pins at once. I suppose I need to manually compare the values against the previous values. I'll give that a try next.

pgrawehr, 2023-10-15:
So, looks good in simulation now, I hope I didn't worse anything :-D

wolfgangmauer, 2023-10-15:
@pgrawehr Tested, works perfect, THX !!!!

closed 2023-10-19 by krwq