dotnet/iot #2137: Raspberry -> SerialPort cannot set a custom baud rate

wolfgangmauer, 2023-09-22:
Try to use a VOTRONIC DC/DC VCC 1212-30

SerialPort.Open throws a exception
```
var votronic = new SerialPort("/dev/ttyUSB0", 1000);
votronic.Open();
```
System.ArgumentException
  HResult=0x80070057
  Nachricht = Value does not fall within the expected range.
  Quelle = System.IO.Ports
  Stapelüberwachung:
   bei System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
   bei System.IO.Ports.SerialPort.Open()
   ....

pgrawehr, 2023-09-23:
Are you sure you need this? I suppose it is really the hardware that fails here. I have never seen a device using a non-standard baudrate bellow 115200.

wolfgangmauer, 2023-09-24:
That would be important. This also works with JS.
```
this.manufacturer = 'Votronic'
this.model = 'VCC 1212-30'
...
this.port = new SerialPort({
      path: connection.port,
      baudRate: **1000**,
      autoOpen: false
    })
```
No hardware failure if using ioctl !

pgrawehr, 2023-09-24:
I can't see any custom exception being thrown there, so it does seem that the ioctl call returns the error (for whateverreason).

The class `System.IO.Ports.Serialport` is not currently part of this repo, so we're not really responsible for it. We're developing a long-term replacement, but that will still need some time. I'll try to reproduce this issue anyway.

krwq, 2023-09-28:
Is it possible this fixes your issue? https://github.com/dotnet/runtime/pull/80534

wolfgangmauer, 2023-09-28:
I use .net 7.0 linux-arm64, System.IO.Ports(5.0.1) came with Iot.Device.Bindings
  ```
<PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net7.0</TargetFramework>
    <ImplicitUsings>true</ImplicitUsings>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="Iot.Device.Bindings" Version="3.0.0" />
    <PackageReference Include="MQTTnet" Version="4.3.1.873" />
    <PackageReference Include="Newtonsoft.Json" Version="13.0.3" />
    <PackageReference Include="System.Configuration.ConfigurationManager" Version="7.0.0" />
    <PackageReference Include="System.Device.Gpio" Version="3.0.0" />
  </ItemGroup>
```
[image]

pgrawehr, 2023-10-01:
@wolfgangmauer You can try to use a newer version of System.IO.Ports anyway. Just reference the newer version in your project directly. When direct and indirect dependencies reference different versions of the same library, .NET will automatically resolve to the latest version, assuming they're backwards compatible (which should absolutely be the case on this one).

wolfgangmauer, 2023-10-07:
@pgrawehr It doesn't work with System.IO.Ports <= 7.0.0 but it works with preview 8.0.0-rc.1.23419.4

pgrawehr, 2023-10-07:
Very good, that sounds like you have a working solution. I guess we can close this here, as there's nothing actionable for us and the bug has been fixed.

closed 2023-10-07 by wolfgangmauer