dotnet/iot #2141: creating multiple devices at the same address with the tca9548a

boondoggler74, 2023-10-03:
**Describe the bug**
Using the tca9548a multiplexer class, I am unable to create two or more devices on different channels at the same address at the same time.  If I attempt to do so without first removing the previously added device, I receive an argument exception. 

This implies that in order for me to be able to use the tca9548a, I must create and remove the device every time i need to use it before any other code creates another device at the same address on a different channel.  Is that the intent of the tca9548a - if so, then this might not be a bug and might be intended behavior.

My understanding is that the purpose of the tca9548a multiplexer is to all us to have devices on different channels at the same address so that we can avoid addressing collisions.  AFAIK, we should be able to switch between channels to interact with the different devices.  Example, I should be able to have an airflow sensor at address 0x61 on channel0 and an mcp4725 dac on channel1 at address 0x61.  Unless I am misinterpreting the intended usage of the tca9548a class, I would not expect that I would be required to remove one device from channel0 before adding the other device to channel1.

I would expect that I could simply create deviceA at address 0x61 on channel0 then create deviceB at address 0x61 on channel1 without exception.  I would expect that before I use deviceA, I switch to channel0 and before I use deviceB I switch to channel1.  It is my understanding this is how the tca9548a is intended to work.

**Steps to reproduce**
Using a C# console app, targeting linux-arm, and rpi4b, standard raspberry pi os, with the I2cBus enabled and Bus 1 configured:
- Create a new I2cBus for BusId 1
- Create a new I2cDevice for the multiplexer at address 0x70
- Create a New Tca9548A passing in the bus and mux device
- Create an I2cDevice at address 0x61 on channel0
- Create an I2CDevice at address 0x61 on channel1
- Expect to receive an System.ArgumentExcetpin: Device with address 0x97 is already open from System.Device.I2c.UnixI2cBus

```csharp
var muxAddress = 0x70;
var dacAddress = 0x61;
var airflowSensorAddress = 0x61;
var i2cBusId = 1;

var bus = I2cBus.Create(i2cBusId);

var muxDevice = I2cDevice.Create(new I2cConnectionSettings(1, muxAddress));
var multiplexer = new Tca9548A(muxDevice, bus, true);
multiplexer[0].CreateDevice(airflowSensorAddress);
multiplexer[1].CreateDevice(dacAddress);
```

**Expected behavior**

I expect a device to be created on channel0 at address 0x61 and another device on channel1 at address 0x61 and that I can call `multiplexer.SelectChannel(MultiplexerChannel.Channel0);` or 
`multiplexer.SelectChannel(MultiplexerChannel.Channel1`); in order to work with deviceA or deviceB without having to remove deviceA or deviceB after each usage

**Actual behavior**

An argument exception is thrown from UnixI2cBus
```
Unhandled exception. System.ArgumentException: Device with address 0x97 is already open. (Parameter 'deviceAddress')
   at System.Device.I2c.UnixI2cBus.<>c.<CreateDevice>b__12_1(Int32 addr, I2cDevice dev)
   at System.Collections.Concurrent.ConcurrentDictionary`2.AddOrUpdate(TKey key, Func`2 addValueFactory, Func`3 updateValueFactory)
   at System.Device.I2c.UnixI2cBus.CreateDevice(Int32 deviceAddress)
   at Iot.Device.Tca954x.Tca9548AChannelBus.CreateDevice(Int32 deviceAddress)
   at Program.<Main>$(String[] args) in C:\source\bvc\iot-samples\dac-sample\Program.cs:line 25
Aborted

```

**Versions used**
- `dotnet --info` on the machine being used to build
```
.NET SDK:
 Version:   7.0.101
 Commit:    bb24aafa11

Runtime Environment:
 OS Name:     Windows
 OS Version:  10.0.22621
 OS Platform: Windows
 RID:         win10-x64
 Base Path:   C:\Program Files\dotnet\sdk\7.0.101\

Host:
  Version:      7.0.1
  Architecture: x64
  Commit:       97203d38ba

.NET SDKs installed:
  7.0.101 [C:\Program Files\dotnet\sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.12 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 7.0.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.12 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 6.0.16 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
  Microsoft.WindowsDesktop.App 6.0.12 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 6.0.16 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
  Microsoft.WindowsDesktop.App 7.0.1 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
```
- target machine is linux-arm rpi4b running raspberry os with a self contained, single file app
- Version of `System.Device.Gpio` 3.0.0
- Version of `Iot.Device.Bindings` 3.0.0
-->

pgrawehr, 2023-10-03:
Ups, you are right. I can already see the error. Due to confusing names in that class, the device is actually created on the master bus instead of the client. I can try to fix it, but unfortunately, I do not have the TCA9548 device, so I cannot do full testing.

boondoggler74, 2023-10-03:
i have some in my lab already wired up, i could test a pre-release nuget package or something

pgrawehr, 2023-10-12:
@boondoggler74 I've created a pull request with what should fix the issue, I believe. Please test when you have time.

boondoggler74, 2023-10-13:
The code in the pr works.

I pulled down the forked repo located at pgrawehr:TCA_DuplicateFix and checked out the branch TCA_DuplicateFix, then I ran the build, after everything compiled locally, I removed my nuget package references from my program, then added assembly references to the debug assemblies that were built in:
- artifacts\obj\Iot.Device.Bindings\Debug\net6.0\Iot.Device.Bindings.dll
- artifacts\obj\System.Device.Gpio\Debug\net6.0\System.Device.Gpio.dll

After that I compiled and published my code with the updated assembly references to the publish dir.  Then I copied it over to the pri4b via winscp and ran it.  It all works.

This code below is the code that runs when my program starts.  You can see that both devices are at the same address but on different channels.  Before I use each device, I simply call mux.SelectChannel(airflowSensorChannnel) or mux.SelectChannel(dacChannel) then interact with the device accordingly.

```csharp
var dacChannel = 1;
var airflowChannel = 0;

var muxAddress = 0x70;
var dacAddress = 0x61;
var airflowSensorAddress = 0x61;
var i2cBusId = 1;

var bus = I2cBus.Create(i2cBusId);
var muxDevice = I2cDevice.Create(new I2cConnectionSettings(i2cBusId, muxAddress));
var mux = new Tca9548A(muxDevice, bus, true);

var dac = mux[dacChannel].CreateDevice(dacAddress);
var airflowSensor = new AirflowSensor(logger, mux[airflowChannel].CreateDevice(airflowSensorAddress));
```

Now that the devices are created I can just switch channels and work with the devices...
`mux.SelectChannel((MultiplexerChannel)dacChannel);`
`mux.SelectChannel((MultiplexerChannel)airflowChannel);`

pgrawehr, 2023-10-13:
@boondoggler74 Thanks for testing, that sounds good. 

Actually, the I2cDevices you obtain above should do the channel selection automatically on each device use. So you should be able to talk to both devices without explicitly calling `SelectChannel` each time. (The bus does not support multi-threading though. So all device reads/writes on the I2C bus must be done on the same thread)

boondoggler74, 2023-10-13:
Thanks for that note.  I did remove my calls to set the channel explitily and then redeployed and tested.  Everything did work as expected.

In my real usage scenario I will be working with the devices in a background worker and also in a web api hosted by the same process.  I will need to write blocking code / critical sections to make sure neither threads are trying to work with the device at the same time.  Thanks for the heads up on the multithreading support.

PR #2146 (2023-10-18): Allow creation of duplicate device addresses for the Tca9548A.
Fixes #2141 

This should fix the use case where multiple devices with the same address are used with a TCA9548A. That's one of the main use cases for this device, after all. I have no such device, so the code is only tested using unit testing. 

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/2146)
closed 2023-10-18 by pgrawehr