dotnet/iot #2157: MCU support

sajjadarashhh, 2023-11-05:
There is any consideration for supporting mcu's like esp32 and... in .net iot?

raffaeler, 2023-11-05:
Hi @sajjadarashhh there is a sister project called `nanoframework`: https://nanoframework.net/
The source repository is here: https://github.com/nanoframework

Many bindings are or can be shared between `dotnet/iot` and `nanoframework` but the resource constraints of the ESP32 do not allow to fully load the full .NET runtime.

@Ellerbach is one of the primary contributors and can tell you far more than me.

pgrawehr, 2023-11-05:
@sajjadarashhh We also have a (prototype) project under samples/ArduinoCsCompiler that can run a limited set of bindings on MCUs like the ESP32. It's far less complete than nanoframework, but on the other hand can technically run the same code as the PC does.

raffaeler, 2023-11-05:
Sorry for forgetting it, @pgrawehr!

sajjadarashhh, 2023-11-06:
I see them but in that projects support is very bad and theres no stable versions of .net for esp32 platform
And documentations is not useful
And samples is not written in latest version of .net nano framework
All of that reason i said make that project awful...
But if we can create .net iot for mcu's we can giving help from .net community and create stable and useful framework

pgrawehr, 2023-11-06:
@sajjadarashhh Please avoid ranting, that doesn't help. 
Can you say what you tried? When I look at [this page](https://docs.nanoframework.net/content/getting-started-guides/getting-started-managed.html), it looks like a step-by-step instruction that's easy to follow. I haven't had time to look into that yet, but certainly will give it a go as well.

Ellerbach, 2023-11-07:
@sajjadarashhh if you are looking for native support of .NET on MCU, please have a look at [.NET nanoFramework](https://github.com/nanoframework). API are aligned with .NET IoT including the bindings.

The goal of .NET nanoFramework is different from the samples/ArduinoCsCompiler you point out. And the step by step page pointed out should help you.

So, depending on what you need to do, one or the other is the proper answer. .NET nanoFramework is more complete and allows different scenarios.

sajjadarashhh, 2023-11-12:
@Ellerbach that nano framework libraries and bindings can be nice if documentations is useful
But for my scenarios that is not complex just i wana to give details from keypad 3x4 or working with Bluetooth of esp32 WROOM that samples in nano framework github is not working and no one help me to solve that problems 
I used nanoframework but i dont understand what happend 
One people in discussions tell me youust be pull up pins but i dont see anything about it in nanoframework documentations 😑

raffaeler, 2023-11-12:
> One people in discussions tell me you must be pull up pins but i dont see anything about it in nanoframework documentations

Pull-ups are resistors providing a stable stable to the HIGH level when a switch or other device is unconnected.
You can setup pullups in two ways:
- adding a physical resistor to the line (this is the preferred way). Depending on the voltage and the environment noise you may want to use roughly a 10K for an ESP32 in an environment with low noise
- using the pull-ups available through only specific pins of the ESP32. You have to look at the ESP32 datasheet because every IC model may provide the functionality on different pins. I never used nanoframework but I could easily find the [SetPinMode](https://github.com/nanoframework/System.Device.Gpio/blob/ec35ddf5c1d630aa198e581121cd93173f348721/System.Device.Gpio/GpioController.cs#L300) function and the [PinMode.InputPullUp](https://github.com/nanoframework/System.Device.Gpio/blob/ec35ddf5c1d630aa198e581121cd93173f348721/System.Device.Gpio/PinMode.cs#L25) enumeration value. Be advised that you cannot just set the pullup without ensuring the ESP32 supports this mode on that pin.

Said that, in addition to the pull-up you may need at least a capacitor to ground so that you realize a RC filter that avoid the [switch bouncing problem](https://www.google.com/search?q=switch+bouncing). The most reliable solution to this physical problem is to use a [schmitt trigger like the 74HC14](https://www.ti.com/lit/ds/symlink/sn74hc14.pdf). You may find a lot of articles explaining this huge topic.

Said that, all of this requires hardware knowledge rather than software documentation. This is probably why you didn't find much in the nanoframework web pages.

Ellerbach, 2023-11-17:
@sajjadarashhh as it seems that quite a lot of beginners are missing some of the basics, I'm starting content for beginners and explain some of the key concepts: https://github.com/nanoframework/nanoframework.github.io/pull/191