dotnet/iot #2181: ILi9341 on Raspberry Pi 4

alberk8, 2023-11-28:
**Describe the bug**

I am trying out the code snippet on the [Readme](https://github.com/dotnet/iot/blob/74f56fae18917dd20379254e76dc21ae1c69e52c/src/devices/Ili934x/README.md) and could not get it to compile.  

**Steps to reproduce**

```csharp
// put sample code here
```

**Expected behavior**
Expect it to run on Raspberry Pi 4.

**Actual behavior**

**Versions used**
dotnet 6.0 SDK
Latest Nuget

Ellerbach, 2023-11-28:
@alberk8 have you tried the sample itself?

https://github.com/dotnet/iot/blob/74f56fae18917dd20379254e76dc21ae1c69e52c/src/devices/Ili934x/samples/Program.cs

All compiles otherwise, we can't merge! So I'll be surprised if it's broken. Maybe the readme is not up to date indeed

alberk8, 2023-11-29:
@Ellerbach,

I managed to get the TFT screen to light up and working but I have this issue with the link that you have provided.  The line causing the issue [this](https://github.com/dotnet/iot/blob/74f56fae18917dd20379254e76dc21ae1c69e52c/src/devices/Ili934x/samples/Program.cs#L31).   Otherwise the FillRect works.

```
Unhandled exception. System.InvalidOperationException: No image factory registered. Call BitmapImage.RegisterImageFactory() with a suitable implementation first. Consult the documentation for further information
   at Iot.Device.Graphics.BitmapImage.VerifyFactoryAvailable()
   at Iot.Device.Graphics.BitmapImage.CreateBitmap(Int32 width, Int32 height, PixelFormat pixelFormat)
   at Iot.Device.Ili934x.Ili9341.GetBackBufferCompatibleImage()
```

Ellerbach, 2023-11-29:
@pgrawehr you may want to have a look at this one

pgrawehr, 2023-11-30:
I'm quite sure I have tested this (and it would be really bad if SkiaSharp wouldn't work on the RPI). I'll give it another try as soon as possible. 

Make sure you call `SkiaSharpAdapter.Register()` at the beginning of your application, that should fix the above exception.

PR #2185 (2023-12-02): Add a missing statement to the documentation
Fixes #2181

The documentation was missing this important instruction. I've verified the binding with the sample under /samples/M5StackRemoteDisplay once again on Windows and on my Raspberry Pi 4 (arm64). Both are working fine for me.

###### Microsoft Reviewers: [Open in CodeFlow](https://portal.fabricbot.ms/api/codeflow?pullrequest=https://github.com/dotnet/iot/pull/2185)
closed 2023-12-02 by joperezr