dotnet/iot #2262: Raspberry Pi 5 -> GPIO Package Can it Support as of Today ????

jldigi40, 2024-01-05:
I have been playing with the current latest 3.10 GPIO Nuget Package and worked on Rasperry Pi 4.

When use on  Pi 5 get exception error ...

1/5/2024 4:49:05 PM :  Cannot open-create Input Device ?....System.IO.IOException: Error while requesting event listener for pin 17, error code: 16
   at System.Device.Gpio.Drivers.LibGpiodDriverEventHandler.SubscribeForEvent(SafeLineHandle pinHandle)

I get same error if change pinnaming or any pin mode setting.

Onlu Open and Close Pin causes no exception

I believe Pi4 and 5 the registers on board a different to drive the 40 pin I/O pins ?

Can I get confirmation Pi5 is not supported as of today using this GPIO ?

Funny think I2C drivers for PI 4 work on PI 5 as use this also, but GPIO is failing.

pgrawehr, 2024-01-05:
Which driver are you using? We did have confirmation that Libgpiod should work on the Raspberry Pi 5. It is expected that the RaspberryPi3Driver is not working, as that uses an interface that is not available on the Pi5. 

Please show a simple example that causes the above error.

krwq, 2024-01-05:
The error message suggests that comes from Libgpiod. @jldigi40 I'm not sure the events are supported on Pi 5, I'd start with regular Read/Write before trying that. Also you might want to try to pass different gpio  chip, I recall the chip might be 1 rather than default 0 (but I might misremember). Here's how you do that: `new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver(gpioChip: 1))`

jldigi40, 2024-01-07:
Hello ALL,

Thanks, quick response, I had a play with recommended using LibGpiodDriver as recommended and if use LibGpiodDriver(gpioChip: 1) I get exception to creating a GpioController Object. If run LibGpiodDriver(gpioChip: 0), get no exception

var controller = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver(gpioChip: 0)); 
 // new LibGpiodDriver(gpioChip: 0) work Pi5
                    
  //// DIO6

  controller.OpenPin(24, PinMode.InputPullUp);

  controller.OpenPin(24).ValueChanged += pin24Change;

Also the event handling like attach handler to 'ValueChanged' works all ok so far

I need to run some real world test simulate inputs and see if reacts correctly..... cross fingers

Just some background, 

I have a PI and an Industrial IO Carrier from 'UniPi' - Product is UniPi 1.1 Lite - https://www.unipi.technology/unipi-1-1-lite-p83

It  has 6 optically isolated inputs (0-24V) and 6 relays - controlled from I2C MCP 23008 chip , this I have working - ie output control ( pi3-4 and now pi 5) End project is a product dev R&D-POC for a Industrial Edge Device Handler.

I need IO to monitor machine inputs the unit will be attached too - ie legacy machinery so need to control some outputs also as well as three separate industry protocols (TCP-IP) based. Basing solution on Blazor for a Web Interface and C# backend for protocol and application Edge IO control.

huesla, 2024-01-13:
For Pi5 you have to use GPIO chip 4 instead of 0, as compared to Pi4. This is because Pi5 has a slightly different GPIO line schema. Use the `gpioinfo` command for more information, it comes with the libgpiod package.

Ellerbach, 2024-01-18:
[Triage] We should document how to properly run it on a RPI5. And creating a RPI5 board class for a start. Later on, we will switch to a dedicated driver, it will work the same but faster.

PRIMETSS, 2024-01-28:
Having similar issues after moving from a working to latest stable SDK & OpenWRT (musl-linux) on RiPi Compute Module 4.
'Segmentation fault' happens when assign the ValueChanged callback.
This project had been running fine with previous version(s) & GPIO buttons were working. 

`_gpioPinButtonUp.ValueChanged += GpioPinButtonUp_ValueChanged`

Using
Dotnet SDK v8.101 (Alpine: dotnet-sdk-8.0.101-linux-musl-arm64)
OpenWRT v23.05.2 YES
	Prev: (GPIO event working [hw buttons])
		Dotnet v8.0.100-rc.2.23502.2
		OpenWRT SNAPSHOT r24111-66f6c20e45

To replicate I build a simple console app, add dependancy `System.Device.Gpio` (v3.1.0, tried right back to 2.2)

Build & Run App on RPi CM4 and get "Segmentation fault" after trying to assign the call back.
[image]

Have also tried the LibGpiodDriver when building the GPIO controller.
`using var controller = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver(0));`

Did a `strace` and get this which might shine some light?

Looks for Library OK:
[image]

Fault:
[image]

Code:
[image]

Remark out the Event Callback assign, and runs as expected.
Has something change recently with assigning callbacks in dotnet or is it a kernal/linux lib issue?

Ta!

pgrawehr, 2024-01-28:
@PRIMETSS That does look like some problem either with the low-level driver or with the registering of the event itself. I quickly checked, and there where no significant changes to the libgpiod driver since at least v2.1. Can you tell which version works? Or is it now failing independent of the System.Device.Gpio library version?

PRIMETSS, 2024-01-28:
Looking at the repo seems there is work underway to better handle libgpiod v1 & v2 (new)
So I added daily build feed to Nuget and tried the latest `<PackageReference Include="System.Device.Gpio" Version="3.2.0-prerelease.24077.2" />`

And tried with V1 & V2 (system is using v1.6)

`var LinuxLibGpioDriver = new LibGpiodDriver(gpioChip: 1, LibGpiodDriverVersion.V1);`
`var LinuxLibGpioDriver = new LibGpiodDriver(gpioChip: 1, LibGpiodDriverVersion.V2);'

Same 'Segmentation fault'

@pgrawehr OK Thanks I will open another issue

krwq, 2024-05-16:
[Triage] We're closing this issue in favor of https://github.com/dotnet/iot/issues/2261 and https://github.com/dotnet/iot/issues/2315 - there is also a workaround for this: using LibGpiodDriver with chip number 4