dotnet/iot #2271: Improve library detection algorithm in LibGpiodDriverFactory

pgrawehr, 2024-01-26:
The current detection algorithm in LibGpidDriver factory is somewhat risky, in that it assumes certain default paths for libraries. A more standard approach would be to just try to use PInvoke and catch the error.

krwq, 2024-04-11:
[Triage] We want to repeat logic in System.DirectoryServices.Protocols in dotnet/runtime: https://github.com/dotnet/runtime/blob/ff3c5314badbb0952bc86325baf12f2dd6255f6f/src/libraries/Common/src/Interop/Linux/OpenLdap/Interop.Ldap.cs#L70-L99

krwq, 2024-04-11:
we should also look at https://github.com/dotnet/iot/issues/2272 as part of this issue as that code might not be needed anymore

krwq, 2025-07-24:
[Triage] We need to change https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L507