dotnet/iot #2280: system.device.gpio is missing NuGet package README file

lyndaidaii, 2024-02-12:
We've noticed that your package on NuGet.org is missing a README file.

  ## Why READMEs are Important
  Our customer research indicates that one of the top problems that package consumers face is insufficient package documentation such as README files. Adding a README file to your package will help users quickly understand what the package is and what it does. Since your README will be the first impression for users when they view your package on NuGet.org, it is crucial for authors to write and include high-quality READMEs for their packages.

  ## Get Started with READMEs

  ### How to Add a README
  If you're new to NuGet READMEs, follow guidance on [how to add one to your package](https://devblogs.microsoft.com/nuget/add-a-readme-to-your-nuget-package/).

  ### How to write high quality of READMEs
  Follow [a blog post](https://devblogs.microsoft.com/nuget/write-a-high-quality-readme-for-nuget-packages/) with README best practices and a template file to help you get started. 

  ## We Value Your Feedback
  We value your feedback. If you encounter any issues or have suggestions, please reply to this issue.

  Thank you for your contribution to the NuGet community.

krwq, 2024-04-11:
@Ellerbach is this fixed by https://github.com/dotnet/iot/pull/2282 ? Can we close it?

Ellerbach, 2024-04-12:
> @Ellerbach is this fixed by #2282 ? Can we close it?

Yes!

closed 2024-04-12 by Ellerbach