dotnet/iot #2283: Add LIS3DSHTR to LIS3DHTR code

JTrotta, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
No problem.

Yes: LIS3DH

**Describe the ideal solution**
It would be nice to use the same library/code to use both. It's just a matter of registry addresses.

Choose the version of LIS during object creation.

krwq, 2024-02-22:
@JTrotta can you find specific differences between those? If it's just the address difference you should be able to reuse existing binding - if the registers are completely different it probably makes sense to keep them separate anyway.

JTrotta, 2024-02-22:
The differences are not related only to the slave address, but to the whole addresses (CTRL1, 2, 3, 4 , etc.). But the usage it's identical. You suggest to create a new device or to generalize the existing one?

pgrawehr, 2024-02-25:
If possible,  we generally try to generalize, e.g. with subclasses. It's also OK to rename namespaces, if needed.

krwq, 2024-03-28:
[Triage] @JTrotta we're not tracking device requests, please open a PR if you're interested in contributing or re-open / create a new issue if you have any more questions

closed 2024-03-28 by krwq