dotnet/iot #2284: logical GPIO

XiaoBaiXJ, 2024-02-26:
My logical GPIO pins are more than 32, but when I run the code and input a number greater than 32, the system reports an error. Why is this happening?

Ellerbach, 2024-02-26:
Can you please give more details on which board you are using, the driver used, and any other information that can help understaing your issue?

XiaoBaiXJ, 2024-02-27:
In an ARM64 environment with .NET 7, attempting to use a GPIO logical pin number greater than 31 results in a warning that the pin is not found
/
[image]

Ellerbach, 2024-02-27:
Can you please give us details? You tried to upload a picture which has not been uploaded.
And also please give details about the hardware, the driver you are using, what is installed

XiaoBaiXJ, 2024-02-28:
[image]
Use .Net7 ARM64

XiaoBaiXJ, 2024-02-28:
If the input GPIO logical port exceeds 31, the system will report an error with a "Segmentation fault".

closed 2024-02-28 by XiaoBaiXJ