dotnet/iot #2290: System.Device.*

Korporal, 2024-03-05:
I'm trying to find out what other packages exist for additional devices, for example timers. Consider theSTM32MP family, these have lots of timers (just as their MCUs do) but I can find no explicit support or discussions about these. 

Any information is appreciated!

Ellerbach, 2024-03-06:
Check out .NET nanoFramework: https://www.nanoFramework.net and https://github.com/nanoframework

It is dedicated to work on MCU, API are aligned with .NET IoT (small differences but 99% of the code is impatible), there most all the Bindings available as well and some specific MCU additional ones.

Ellerbach, 2024-03-21:
[Triage] We can only add support what runs .NET natively. On top of the previous answer, you can look at https://github.com/dotnet/iot/tree/main/src/devices/Arduino, that can also be something that fits your needs.

So closing this issue. Feel free to reopen in case.

closed 2024-03-21 by Ellerbach