dotnet/iot #2293: Iot.Device.Media VideoDevice  Capture creates corrupt images with Raspberry PI Camera module 3

chrisgilliam72, 2024-03-16:
I have been successfully using the code sample I found here:
https://neilsnotes.net/Software/Coding/dotnetPiCam.html
To capture images from my Raspberry PI  with camera module v2.1

I recently bought a new camera module v3.  The same code runs but when I try to view the captured image it , the image appears to be corrupt  : Error interpreting jpeg image file (not a jpeg file starts with 0x00 0x00)

raffaeler, 2024-03-18:
Hi @chrisgilliam72, could you please try the [`Camera` binding](https://github.com/dotnet/iot/tree/main/src/devices/Camera) instead?

The most probable reason for the `VideoDevice` not working anymore is because the Rasberry foundation deprecated the "legacy stack" for the newer "libcamera".
You can still revert to the first one in the raspi-config, but since I created the `Camera`, you can work with both stacks.

chrisgilliam72, 2024-03-18:
Thanks for the response . Does the Camera device have something like the NewImageBufferReadyEvent I can hook onto ?

raffaeler, 2024-03-18:
@chrisgilliam72 The Camera binding runs the official Raspberry PI `libcamera-vid` utility and takes the video stream back.
This means that you get a continuous `Stream` flow that you can send to a web page. On the binding side, I could not expose that event because the stream content depends on the options you pass to the tool.

The code samples just show how to acquire the stream, but I successfully streamed the `byte[]` flow to the browser (which needs a video player javascript library).
I did this in the past with the H264 stream because it is low-bandwidth, but I had to manually separate the individual frames because the javascript library required the video frame separation.
If you don't want to mess with H264, you can change the codec to MJPEG (I believe this was the `VideoDevice` default codec) but I don't know if the frame separation is still needed (it should be easier to implement.

I planned to post the video frame splitting code, but it will take some time.

raffaeler, 2024-03-20:
> I have checked those camera binding and it looks like it does not work also on RaspBerry Pi 5. Even listing cameras. I have couple of them it is working on RaspBerry Pi 4 but not on 5th.

@Woselko May I ask you to open a separate issue with detailed information?
The `camera` binding runs a separate process with the official tool called `libcamera-vid`. If the `libcamera-vid` works from the terminal and does not work from the binding, there is either a misconfiguration or a bug.
But if it does not work from the terminal, there is something wrong that has nothing to do with this repo.

Thank you

raffaeler, 2024-03-20:
@chrisgilliam72 You should be able to do this adding this line:
```csharp
var builder = new CommandOptionsBuilder()
// ...
    .With(CommandOptionAndValue.Create(Command.Nopreview));
```

Woselko, 2024-03-21:
> @Woselko May I ask you to open a separate issue with detailed information? The `camera` binding runs a separate process with the official tool called `libcamera-vid`. If the `libcamera-vid` works [...]

Exactly, it works from terminal but now from the binding.
I will open a separate bug.

Can someone explain me why there is so much differences between raspberry pi 4 and raspberry pi 5?
Only few devices from this package works from me.

raffaeler, 2024-03-21:
> Exactly, it works from terminal but now from the binding. I will open a separate bug.

Thank you, please detail the specific error.

> Can someone explain me why there is so much differences between raspberry pi 4 and raspberry pi 5? Only few devices from this package works from me.

The Broadcom chipset is different. Did you find a commn root for these issues?
It may depend on the `libgpiod` vs `sysfs` driver usage (there are relevant issues on the Pi5 and `libgpiod`).
In any case, please open one or more issues on this.

joperezr, 2024-03-21:
@Woselko can you clarify which devices stopped working for you in the Pi5? Essentially we are trying to understand if these are devices using I2c communication or GPIO, in order to try to narrow down the issue.

Woselko, 2024-03-25:
@joperezr here I made an issue: https://github.com/dotnet/iot/issues/2266

raffaeler, 2024-03-25:
@Woselko 
- with regards to the request #2266 I commented there.

- You are doing a great work on your repo. Should you want to contribute in this repo as well, you are more than welcome :-)

- Camera binding (this issue). Please provide more detailed info about the problems if you still can't make it work the Camera binding on the RPi5. As I wrote in previous comments, the video stream is acquired by the native utilities, therefore I do not expect the RPi5 to be the culprit here. The binding just wraps the utilities in order to obtain the stream from the Camera.
If instead you could make it work, please close this issue.

Thank you!

Woselko, 2024-03-28:
@chrisgilliam72 check my LibCamera C# wrapper in my RaspberryPi 5 repo: https://github.com/Woselko/DotnetRaspBerryPi5Patterns
there is project called RaspCameraLibrary which is libcamera wrapper and I was able to stream some video locally on raspberry pi 5 server with CameraStreamServer project.
[image]
im still new to raspberry platform so it can be some mistakes :)

closed 2024-04-06 by dotnet-policy-service[bot]