dotnet/iot #2295: Does the library support TRF7970A?

huangmagyc, 2024-03-17:
In a new case I received, the customer's equipment is equipped with a TRF7970A sensor. Does the current iot library support TRF7970A?

pgrawehr, 2024-03-18:
@Ellerbach Looks like an RFC chip, so you might know.

Ellerbach, 2024-03-18:
Ther is no support for TRF7970A or TRF7970A EVM.
That said, you have multiple examples of NFC readers in the repository. And all the card like Mifare, Ultralight, NDEF on top of those and Credit Card will work if you implement the proper interface. See https://github.com/dotnet/iot/tree/main/src/devices/Card

For the supported ones:

* https://github.com/dotnet/iot/tree/main/src/devices/Pn5180
* https://github.com/dotnet/iot/tree/main/src/devices/Pn532
* https://github.com/dotnet/iot/tree/main/src/devices/Mfrc522

The complexity of implementation will highly vary depending on the chip maker. Some provide high level functions, some others not really. In all cases, they should provide a proper documentation to be able to implement a driver.

Happy to support with question and we love PR :-)

krwq, 2024-03-28:
[Triage] We don't keep requests for new bindings, please open a PR if you're interested in adding support - you can re-open or create a new issue if you have any further questions

closed 2024-03-28 by krwq