dotnet/iot #2296: FT232H Cannot register for pin value change events

IamRewt, 2024-03-20:
**Describe the bug**

When using an FT232H board on windows, attempting to call GpioController.RegisterCallbackForPinValueChangedEvent() results in a 'NotImplementedException'.

The exact message is: "System.NotImplementedException: 'The method or operation is not implemented.'"

**Steps to reproduce**

```csharp
 Ft232HDevice ft232h = new(FtCommon.GetDevices()[0]);
 var gpio = ft232h.CreateGpioController();
 var pin = 0;

 gpio.OpenPin(pin);
 gpio.SetPinMode(pin, PinMode.Input);
 gpio.RegisterCallbackForPinValueChangedEvent(pin, PinEventTypes.Rising, HandleRisingEvent); // Exception thrown here.

// the method being registered
  private void HandleRisingEvent(object sender, PinValueChangedEventArgs pinValueChangedEventArgs)
  {
      Console.WriteLine("Sensor changed from Low to High!");
  }
```

**Expected behavior**

When using a gpio controller on say a raspberry pi, registering the method in the same way works fine and events are caught.

For example, this code runs fine on a raspberry pi.
```csharp
var gpio = new GpioController();
gpio.OpenPin(4);
gpio.SetPinMode(4, PinMode.Input);
gpio.RegisterCallbackForPinValueChangedEvent(pin, PinEventTypes.Rising, HandleRisingEvent);
```

**Actual behavior**

A Not Implemented exception is thrown.

**Versions used**
.Net Version:  8.0.3
IOT Device Binding Version: 3.1.0.
System Device GPIO Version: 3.1.0

Ellerbach, 2024-03-21:
Hi, this is the actual correct behavior, this is not implemented. See https://github.com/dotnet/iot/blob/ab3f910a76568d8a0c234aee0227c65705729da8/src/devices/Ft232H/Ft232HGpio.cs#L220

If you want to implement it, you can with your own logic.

krwq, 2024-03-28:
[Triage] we're in favor of PRs than bug reports for those kinds of issues - there is no straight forward way to support this natively and we'd need to emulate this behavior

closed 2024-03-28 by krwq