dotnet/iot #2297: Update Raspberry Pi samples suggesting modifying config.txt settings

ChayoteJarocho, 2024-03-21:
The latest available version of Raspberry Pi OS (Bookworm) has changed the location of the config.txt file where you can enable the SPI hardware interface. It used to be in `/boot/config.txt`, and now it is located in `/boot/firmware/config.txt`. Also, the setting went through different names, but the latest name is `dtparam=spi=on`.

Some of the sample docs still have the old instructions and need updating. For example, the nRF24L01 sample: https://github.com/dotnet/iot/blob/main/src/devices/README.md#important

pgrawehr, 2024-03-21:
When was that change made and where is it documented?  I set up a version of bookworm about a month ago and there everything was thr same as before.

pgrawehr, 2024-04-03:
I looked at it and can confirm the change. The only change appears to be the place of the file, though. Also, there's a file in the original place that contains the information to look in the new place - people should be able to understand that.

Ellerbach, 2024-04-04:
[Triage] If you are willing to adjust the documentation in the repository, feel free. In that case, please make sure you are clear about the fact that depending on the OS version, it's one or the other. The params should be already good on what we've been checking. We ❤️ PR so looking forward to reviewing it 😉

krwq, 2024-04-11:
[Triage] on top of docs we need to update Board class config reading logic as part of this