dotnet/iot #2300: Unable to set pin mode on pi5

IamRewt, 2024-03-24:
**Describe the bug**

After updating to version 3.1.0, I am unable to read or set modes on gpio pins on the raspberry pi 5. The library works properly when reverting to 3.0.0.
 
Error Message: Unhandled exception. System.IO.IOException: Error setting pin mode, for pin: 4, error: 16.
**Steps to reproduce**

```csharp
 var gpio = new GpioController();
 gpio.OpenPin(4);
 gpio.SetPinMode(4, PinMode.Input); <-- Throws error.
 Console.WriteLine(gpio.Read(4));
```

**Expected behavior**

The pin should set and be readable.

**Actual behavior**

The pin does not get set and the error message is thrown.

**Versions used**

pgrawehr, 2024-03-25:
That's kind of expected, since we do not have full support for the pi 5 yet. You need to manually select the driver (either libgpiod or sysfs)

IamRewt, 2024-03-25:
There may have been a regression, related to the pi 5, with version 3.1.0 as the gpio seems to function normally in 3.0.0. The above code runs without error and properly reads the pins. I haven't tested all of the gpio pins but have tried several of them without issue.

Given the 5 is still fairly new, I can go back to testing with a 4 for now.

closed 2024-04-06 by dotnet-policy-service[bot]