dotnet/iot #2315: Implement Raspberry5Driver and board

krwq, 2024-05-16:
We should investigate if it makes sense to implement Raspberry 5 driver and board. We should investigate:
- perf differences
- other features between our libgpiod implementation and RPi4 direct implementation

and do the same for RPi5 and figure out if it makes sense to implement it. If the perf is similar we might want to decide to not do this.

Ref: https://github.com/dotnet/iot/issues/2262

Ellerbach, 2024-06-06:
[Triage] Removing the untriage and commenting for the bot not to add it again :-)

gengzupwnz, 2025-03-07:
DHT22 is not working with LibGpiodDriver. But python library works good. Tried different approaches but no luck.

pgrawehr, 2025-03-09:
@gengzupwnz Unfortunately, the DHT22 is very timing sensitive. It seems our driver has some problems with the increased CPU speed of the RPI5. You need to look into the code and try to adjust the timings in the `DhtBase` class (maybe play around with the maximum value of the `_loopCount` variable).

tscholze, 2025-04-25:
If I want to use GPIO on a Raspberry Pi, you would recommend using the "old" Raspberry Pi 4 for it - if not the extra performance of the Pi 5 is required?

raffaeler, 2025-04-25:
We had feedback from users who are already using the GPIOs on the Pi5. At the very end this library accesses the hardware using the native library libgpiod (or sysfs driver which is not the recommended approach).

We had issues on the first batch of the PI5 using a different chip number, but the Raspberry Foundation fixed that.

We opened this thread to improve future versions which may leverage the new PI5 chipset.

In other words, you can pick whatever you prefer.