dotnet/iot #2317: Add support for SPI devices that use separate GPIO outputs for Data/Command and Reset

AdamJSchofield, 2024-05-20:
I am looking to add support for a new SSD1309 device that shares a lot of functionality with [the other SSD13xx](https://github.com/dotnet/iot/tree/main/src/devices/Ssd13xx) devices. However, the existing implementations only use I2C and a "control byte" for Data/Command switching.

I would like to extend the SSD13xx implementation to support SPI with additional GPIO pins for controlling DC and RST. Ideally these pins would be specified via Board.CreateSpiDevice or SpiConnectionSettings.

**The device I am using:** https://www.waveshare.com/1.51inch-transparent-oled.htm
**SSD1309 Datasheet:** https://www.crystalfontz.com/controllers/datasheet-viewer.php?id=342
**WIP PR:** https://github.com/dotnet/iot/pull/2316

Even with hacky work-arounds I have been unable to get this device working. I have validated my device and pin configuration with examples from the [Luma.OLED](https://github.com/rm-hull/luma.oled) and have some other work in progress to port over as much as I can from there to have a working baseline.

I'm looking to start some conversation about what an implementation like this may entail. Is it too specific to be supported by the lower-level SpiDevice? Is bit-banging support a possibility? I'm also interested in opinions on temporary work arounds involving interop with working python code (unrelated to this project I suppose).

AdamJSchofield, 2024-05-20:
And I'll add a full disclosure that I am fairly inexperienced with IoT things in general but I have spent several days in the weeds on this so I'm fairly comfortable with the basics of SPI. But for example, I just learned the difference between hardware/software SPI. I suspect that my device doesn't use hardware SPI and using [software SPI](https://github.com/dotnet/iot/tree/main/src/devices/SoftwareSpi) may resolve some of the issues with my demo project.

joperezr, 2024-05-30:
Hey @AdamJSchofield, thank you so much for your contribution 😃. I noticed you already sent out a  PR, we'll take a look soon to be able to get it in.

AdamJSchofield, 2024-05-30:
Thanks for taking a look! As I mentioned on the [draft PR](https://github.com/dotnet/iot/pull/2316#issuecomment-2140143727), I've worked out some issues and am just polishing up my branch for review. I will probably close this issue now that I've got things figured out and keep discussion on the PR. Unless you prefer an issue for every PR.

AdamJSchofield, 2024-05-30:
Closing because I've worked through most of the issues I was having. Will mark the PR as ready for review in the next few days.

closed 2024-05-30 by AdamJSchofield