dotnet/iot #2320: If using gpio on an arm Linux system. How do I make things displayed on the interface?

bbhxwl, 2024-06-04:
If using gpio on an arm Linux system. How do I make things displayed on the interface?

How to display the software on the screen?

pgrawehr, 2024-06-04:
This library does not contain any classical UI components. It contains drivers for small displays, but nothing to display windows or buttons. You need to use another library for that (eg Avalonia) or just the console.

logicaloud, 2024-06-05:
Funny enough, I have just completed a first version of a browser UI with Docker container backend tailored around the library (running on Raspberry Pi 64-bit). This was/is a fun side project for still ongoing experimentation with the device model (see discussion here: https://github.com/dotnet/iot/issues/2074). The UI allows for designing no-code/low code logic around the devices and you can see results, but if you are after some kind of SCADA interface with LEDs on screen, etc., then this is not it. The UI and backend are among the free modules of a larger commercial package, and I'm not sure what the right etiquette is for providing links. If there is any interest, then I can hop on Discord to discuss.