dotnet/iot #2327: SPI communication on RPI5 is possibly broken

pgrawehr, 2024-06-21:
**Describe the bug**

According to https://stackoverflow.com/questions/78623887/spi-control-on-raspberry-pi-5-using-vs2022-net-4-8 SPI communication on a Raspberry Pi 5 doesn't work.

**Steps to reproduce**

See linked issue.

**Expected behavior**

SPI bus works

**Actual behavior**

No SPI clock seen

**Versions used**

latest (I suppose)

rjhind, 2024-09-19:
I'm successfully using spi0 on a raspberry pi 5 without issue.

`var spiSettings = new SpiConnectionSettings(0, 0) {
    ClockFrequency = 10000000,
};
using var spiDevice = SpiDevice.Create(spiSettings);
`

pgrawehr, 2024-09-21:
@rjhind Thanks for verification. I'm closing this ticket then until we have further details as to why it shouldn't work (sometimes?)

closed 2024-09-21 by pgrawehr