dotnet/iot #2328: ButtonBase.IsPressed not properly initialized

RevZero, 2024-06-26:
**Describe the bug**

After initializing a GpioButton, the IsPressed property is always false, even if the logical state (depending on active low/high configuration) of the pin is true.

**Steps to reproduce**
This examples assumes pin 16 being externally pulled up, having a switch connected which is currently pulling the input low.
```csharp
var gpio = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver());
var input = new GpioButton(16, true, false, gpio, false);
Assert.That(input.IsPressed); // This fails but IsPressed should be true, because the input is low active and being pulled down.
var pin = gpio.OpenPin(16, PinMode.InputPullUp);
Assert.That(pin.Read() == PinValue.Low); // Passes as expected
input.IsPressed = pin.Read() == PinValue.Low;  // Manually set the initial state as a workaround.
```

**Expected behavior**
IsPressed should be initialized according to the initial pin state instead after the first transition.

**Versions used**
- net8.0 self-contained
- Version of `System.Device.Gpio` package: 3.1.0

krwq, 2024-06-27:
[Triage] @RevZero can you try with both LibGpiodDriver, SysFsDriver and RaspberryPi3Driver directly and see if that makes any difference? If you haven't done yet please make sure you have installed libgpiod (with i.e. apt)

Timothy-Lee-Grant, 2026-10-02:
I'd like to pick this up if it's still wanted. On `main`, `GpioButton` never reads the pin's current level, so `IsPressed` stays false until the first edge. Furthermore, with debounce enabled, `HandleButtonReleased` returns early when `!IsPressed`, so the first release of a button held at startup is swallowed (no `ButtonUp`/`Press`).

I have a small change ready: after registering the edge callback, read the pin once and set `IsPressed` from the active level (Low for pull-up, High for pull-down), without raising events. Tests use `MockableGpioDriver`; four fail on `main` and pass with the change, including the swallowed release. But I have two questions that I would like input on: 

1. Timing: #1715 suggests the pull-up may not have settled right after `OpenPin`. Is an immediate read alright, or do you think it would be better to have a short settle delay, or reading lazily on first subscription (#1715's proposed fix)?

2. Ordering: I read after registering the callback, so an edge between the two can't be missed entirely, but there is still a tiny window where a callback could be overwritten by the initial read, which only a lock in `ButtonBase` would close. Is that small window acceptable here?

I've looked at #2608 including the new `TimeProvider` commit: it doesn't change the initial-state behavior, and my change only touches `GpioButton.cs` and the test project, so the two should combine with at most a one-line merge in `Button.Tests.csproj`. I'm happy to rebase on top of it once it's in. @raffaeler, OK for me to take this?

raffaeler, 2026-10-02:
Hi @Timothy-Lee-Grant great points.
About point 1, I am afraid it can also depends on the board. We should definitely avoid send bogus notifications but delays usually cause a lot of issues (very often when running tests as you have seen in the recent PR).

On point 2, it is very use-case depent. Not entirely sure how to solve this but would be nice

You are more than welcome to submit a PR, just be aware that @pgrawehr is working on the other PR and avoid collisions.

What do you think @pgrawehr ?

Timothy-Lee-Grant, 2026-10-03:
Awesome! I'll keep the immediate read with no delay and no events, and note the ordering window in the PR as a known limitation. My change doesn't touch `ButtonBase.cs`; I've checked it on top of #2608's latest commit and all Button tests pass together (the only overlap is one line in `Button.Tests.csproj`). I'll open the PR against `main` and rebase after #2608 merges, unless @pgrawehr would prefer a different order.

closed 2026-10-08 by pgrawehr