dotnet/iot #2334: CONTRIBUTING documentation is way out of date

jdbruner, 2024-07-25:
Instructions for contributors at ```Documentation/CONTRIBUTING.md``` recommend using Visual Studio 2019 on Windows and .NET 6 preview (and even include links to RC preview distributions of .NET 6). At a minimum, this should refer to VS2022 and at least mention using VSCode. It should also point to at least the .NET 6 LTS release; however, with the deprecation of .NET 6 in November, perhaps it is time to encourage people to use .NET 8 instead.

raffaeler, 2024-07-25:
You are totally right. Are you willing to create a PR for this document?
Thanks!

jdbruner, 2024-07-25:
Sure, I'm willing to create a PR, but what should it say about .NET 6 / .NET 8 ?

joperezr, 2024-07-25:
Thank you so much for letting us know and  for volunteering. Regarding .NET 6 vs .NET 8, we currently build the repo using .NET 7 tooling but have a PR coming to  bump that to .NET 8, so in terms of recommendation, we would like to mention that .NET 8 SDK should be installed (or latest VS which would include if developing in Windows). In terms of what framework we target, we currently target .NET 6, but that is going out of support in November, so we also have a PR coming soon-ish that will bump the target framework as well to ensure we target supported platforms only (which echoes what other dotnet repos like dotnet/runtime does).

Please don't hesitate to ask other questions, and let us know if there is any other information we can provide to help 😃

PR #2336 (2024-08-08): update obsolete references and mention VSCode
Fixes #2334 
Replace obsolete links to .NET 6 preview with links to download page for all .NET versions (and adjust text accordingly). Mention (and add download link for) VSCode.
 ###### Microsoft Reviewers: [Open in CodeFlow](https://microsoft.github.io/open-pr/?codeflow=https://github.com/dotnet/iot/pull/2336)
closed 2024-08-08 by krwq