dotnet/iot #2349: Question: FT232H with UWP?

myokeeh, 2024-08-26:
I have an app that for several reasons associated with Microsoft decisions, I cannot fully release as a WinUI3 / WindowsAppSDK app.

Is there a way to allow a UWP app to work with what's documented here: https://github.com/dotnet/iot/tree/main/src/devices/Ft232H?

I've tried the sample code, and it works on a WinUI3 test app, but something's not allowing the UWP test app to properly connect to the FT232H. It sees a device, but that's about as far as it gets.

pgrawehr, 2024-08-26:
Can you explain your problem a bit clearer? You write "I've tried the sample code, and it works on a WinUI3 test app" and also " something's not allowing the UWP test app to properly connect to the FT232H". That is contradictory. Can you show what code you used and what did and what did not work?

myokeeh, 2024-08-26:
```

var devices = FtCommon.GetDevices();
Console.WriteLine($"{devices.Count} available device(s)");
foreach (var device in devices)
{
    Console.WriteLine($"  {device.Description}");
    Console.WriteLine($"    Flags: {device.Flags}");
    Console.WriteLine($"    Id: {device.Id}");
    Console.WriteLine($"    LocId: {device.LocId}");
    Console.WriteLine($"    Serial number: {device.SerialNumber}");
    Console.WriteLine($"    Type: {device.Type}");
}

if (devices.Count == 0)
{
    Console.WriteLine("No device connected");
    return;
}

```

The above code works fine for a WinUI3 / WindowsAppSDK app. It does not work for a UWP app.

Specifically, the device type reported is "UnknownDevice" with the UWP app vs. "FT232H" with the WinUI3 / WindowsAppSDK.

Ellerbach, 2024-08-26:
As far as I remember, UWP requires to declare that you require to access specific hardware. For this, you need to follow the documentation: https://learn.microsoft.com/en-us/windows/uwp/packaging/app-capability-declarations
You may even need to specific USB vendor ID and product ID.

myokeeh, 2024-08-27:
```
  <Capabilities>
    <Capability Name="internetClient" />
    <iot:Capability Name="lowLevelDevices"/>
    <DeviceCapability Name="lowLevel"/>
  </Capabilities>
```
I already had the above capabilities defined (prior to posting here). I also tried `runFullTrust` but didn't make a difference.

First, is there a way to be more definite if this is supposed to work on UWP? Second, if it is, what specific things do I need? I feel there's too many variables.

myokeeh, 2024-09-16:
I'm hopeful maybe this latest news helps in making this all work together.

https://devblogs.microsoft.com/ifdef-windows/preview-uwp-support-for-dotnet-9-native-aot/

myokeeh, 2024-09-18:
I've tried the following and still can't get it to work:

```
		<DeviceCapability Name="usb">
			<Device Id="vidpid:0403 6014">
				<Function Type="classId:ff * *"/>
				<Function Type="name:vendorSpecific"/>
			</Device>
		</DeviceCapability>
```

krwq, 2024-10-10:
[Triage] @myokeeh At the moment, we don't have any immediate plans to look into better support for UWP as our current focus is on other priorities. We believe the issue is about the permissions and you'll need to explore this further to get this to work. Sorry we couldn't be of any more help and we'll close it since it's not actionable for us. If you have any findings please share it with us so that we can point other people with similar problems. Feel free to re-open if you have any more findings.

closed 2024-10-10 by krwq

Ellerbach, 2024-10-10:
@myokeeh one last thing to try, can you try this:

```json
    <DeviceCapability Name="usb">
      <Device Id="vidpid:0403&amp;6014">
        <Function Type="classId:ff * *"/>
        <Function Type="name:vendorSpecific" />
      </Device>
    </DeviceCapability>
```

The difference is in the device ID, it seems that the `&` is needed and was missing in your definition

myokeeh, 2024-10-11:
@Ellerbach, does not work with the ampersand.

```
Validation error. error C00CE169: App manifest validation error: The app manifest must be valid as per schema: Line 44, Column 15, Reason: 'vidpid:0403&6014' violates pattern constraint of 'any|vidpid:[0-9a-fA-F]{4} [0-9a-fA-F]{4}( (usb|bluetooth))?|model:[^;]{1,512};.{1,512}'. The attribute 'Id' with value 'vidpid:0403&6014' failed to parse.	FT232H Test	C:\source\repos\FT232H Test\FT232H Test\bin\x64\Debug\AppxManifest.xml		

```

Ellerbach, 2024-10-11:
@myokeeh then I really don't have any other idea!