dotnet/iot #2363: Are there compatible boards that integrate PWM pins?

phiteo1, 2024-11-15:
Hello, everyone. 
I need to drive a device via PWM using, if possible, an FT* compatible board. 
However, of all the compatible boards, none seem to have a PWM output pin. Are there any PWM expansions that it is possible to drive from one of these boards?
N.B: I do not have an arduino or raspberry type board, but a mini PC. Consequently, the need for an FT* board is to have GPIOs.

pgrawehr, 2024-11-16:
Hi!

There are various solutions for this, the best solution depends on your detailed needs, though. What exactly are you trying to drive through PWM? If it's a motor, you might go for something like a [PCA9685 motor driver](https://github.com/dotnet/iot/blob/main/src/devices/Pca9685/README.md), as you additionally need a power amplifier anyway. FT* boards do support I2C interfaces.

If you only need to drive a set of LEDs, you can also use an Arduino or ESP32 in firmata mode, then they work similar to an FT* board, except that they also include PWM support and have many more pins.

raffaeler, 2024-11-16:
I also strongly suggest an external board.
Look at this one: https://www.aliexpress.com/item/1005006486594447.html
The advantages are:
- completely decouple the motor (or other) power from the RPi power
- 16 channels
- 12 bit resolution. As far as I know the pwm resolution of the RPi is 8 bit, but I may be wrong.

phiteo1, 2024-11-16:
> Hi!
> 
> There are various solutions for this, the best solution depends on your detailed needs, though. What exactly are you trying to drive through PWM? If it's a motor, you might go for something like a [PCA9685 motor driver](https://github.com/dotnet/iot/blob/main/src/devices/Pca9685/README.md), as you additionally need a power amplifier anyway. FT* boards do support I2C interfaces.
>  Yes
> If you only need to drive a set of LEDs, you can also use an Arduino or ESP32 in firmata mode, then they work similar to an FT* board, except that they also include PWM support and have many more pins.

phiteo1, 2024-11-16:
Hi

I Need to drive a servo motori. 
So, i can use the I2C ti drive the PCA9686 but i dont understand how to use the driver. If i not have direct access to the PIN in what way i can create new pwm PIN from?

raffaeler, 2024-11-16:
The easiest solution may be to use a small and cheap ESP32 board that you can program using Arduino.
You basically can use this board as a bridge between your mini-PC and the PWM board.
The ESP32 boards have the serial port and you can "forward" the command to the PWM.

The additional value of this solution is that you can check any feedback (like the end position of the motor) in true real-time.
Real-time operations from a PC are never real-time :-)

phiteo1, 2024-11-16:
Thank you.
I thinking about the ft232 because i found this usefull library and the drivers works without any problem. I can use the Arduino and .NET togheter?
Anyway, this code can be a right way to do It?

using Ft232HDevice ft232h = new Ft232HDevice();
I2cBus i2cBus = ft232h.CreateOrGetI2cBus();

const int pca9685Address = 0x40;

using Pca9685 pca9685 = new Pca9685(i2cBus.CreateDevice(pca9685Address), pwmFrequency: 50);

closed 2024-11-16 by phiteo1

phiteo1, 2024-11-16:
There exists the pca9685 class inside the IoT.Device.Pwm.
I saw It in a comment above
https://github.com/dotnet/iot/blob/main/src/devices/Pca9685/README.md

phiteo1, 2024-11-16:
This code can be a right way to do It?

using Ft232HDevice ft232h = new Ft232HDevice();
I2cBus i2cBus = ft232h.CreateOrGetI2cBus();

const int pca9685Address = 0x40;

using Pca9685 pca9685 = new Pca9685(i2cBus.CreateDevice(pca9685Address), pwmFrequency: 50);

phiteo1, 2024-11-16:
> For the nanoframework, I am not sure about the compatibility as the ESP32 is a very constrained device and not every binding available in this repo can run on small devices. The nanoframework devices repo is here: https://github.com/nanoframework/nanoFramework.IoT.Device
>  It works for and x86? because i cant use ARM

However if i want to use the ft232 did It possibile in this way?

raffaeler, 2024-11-16:
> However if i want to use the ft232 did It possibile in this way?

I never used the PCA9685 through the FT232H, but in theory yes.
This is also something that @Ellerbach can know as he authored the FTXXX bindings.

phiteo1, 2024-11-16:
> For the nanoframework, I am not sure about the compatibility as the ESP32 is a very constrained device and not every binding available in this repo can run on small devices. The nanoframework devices repo is here: https://github.com/nanoframework/nanoFramework.IoT.Device
 
The nanoFramework works with and x86 arch? Cause i can run my service on a arm and this Is the reason because i Need to use a mini pc

raffaeler, 2024-11-16:
It does not make sense to use nanoframework on a pc.
Did you try the library from this repo on the pc?

phiteo1, 2024-11-16:
> It does not make sense to use nanoframework on a pc. Did you try the library from this repo on the pc?

Yes I compile the example to use the ft232 Board for Reading a sensori.
Also, in the repo first pags its be wrote that in can bes used with PC if you use the device like the ft232

raffaeler, 2024-11-16:
IMO you should be able to use it for that goal.
I usually tend to avoid Operating Systems in emebedded scenarios as they have a larger attack surface and therefore requires to be updated frequently.

phiteo1, 2024-11-16:
> IMO you should be able to use it for that goal. I usually tend to avoid Operating Systems in emebedded scenarios as they have a larger attack surface and therefore requires to be updated frequently.

Yes, i agree with you.
But the costumer software run only on x86 arch. So i have to adapt to it

krwq, 2024-11-19:
> This code can be a right way to do It?
> 
> using Ft232HDevice ft232h = new Ft232HDevice(); I2cBus i2cBus = ft232h.CreateOrGetI2cBus();
> 
> const int pca9685Address = 0x40;
> 
> using Pca9685 pca9685 = new Pca9685(i2cBus.CreateDevice(pca9685Address), pwmFrequency: 50);

This should work - well - at least in theory - I haven't tried exactly this combination but I never had any issues with I2C devices with FT232 on Windows x64. Having said that I'm not sure if we've ever tested FT232 on x86 so I'd recommend trying on x64 if you're seeing any issues then if that works then possibly it might be a problem with our interop layer which might need a fix for x86 and I'm sure one of us will help you debug and unblock you if needed. Please create a new issue if you see any problems with that code on x86