dotnet/iot #2364: Typo in the Documentation

mikulik86, 2024-11-16:
Hi, i think you have a typo in this article: https://github.com/dotnet/iot/blob/main/Documentation/raspi-pwm.md#enabling-hardware-pwm

In the note there is supposed to be 

> Adjust the previous line to be `sudo nano /boot/config.txt`

istead of 

> Adjust the previous line to be `sudo nano /boot/firmware/config.txt`

raffaeler, 2024-11-16:
Good catch! This changed right before June 2023 according to the official documentation:
https://www.raspberrypi.com/documentation/computers/config_txt.html

> Instead of the [BIOS](https://en.wikipedia.org/wiki/BIOS) found on a conventional PC, Raspberry Pi devices use a configuration file called config.txt. The GPU reads config.txt before the Arm CPU and Linux initialise. Raspberry Pi OS looks for this file in the boot partition, located at /boot/firmware/.
>
> Note: 
> Prior to Raspberry Pi OS Bookworm, Raspberry Pi OS stored the boot partition at /boot/.

I believe it's worth to mention both.
Are you interested in submitting a pull request for this fix?

krwq, 2025-01-23:
[Triage] `/boot/firmware/config.txt` is correct and `/boot/config.txt` is the old path. We also have a note mentioning that in the README. Closing as done

closed 2025-01-23 by krwq