dotnet/iot #2372: What's the meaning of error code in the Exception?

AkazawaYun, 2025-01-06:
Error while requesting event listener for pin 14, error code: 517
Error while requesting event listener for pin 0, error code: 22
Error while reading value from pin: 12, error: 1

                          ↑↑↑↑↑↑↑↑↑↑↑↑↑ 

I can't find the reason or details of the error code in anywhere on the internet...What does the '517', '22' , '1'  mean , and how can I fix it ?

https://learn.microsoft.com/zh-cn/dotnet/iot/
https://www.nuget.org/packages/System.Device.Gpio

raffaeler, 2025-01-06:
The error comes from the underlying native library and is propagated to the exception. I assume you are using `libgpiod` and not `sysfs` but they probably share the same error codes.

According to the `libgpiod` [header file](https://github.com/brgl/libgpiod/blob/master/include/gpiod.h#L34): 
> It's not practical to list all possible error codes for every function as they propagate errors from the underlying libc functions.

This means that they use the `libc` error codes that you can find here:
https://www.gnu.org/software/libc/manual/html_node/Error-Codes.html
https://www.thegeekstuff.com/2010/10/linux-error-codes/

In your case:
- Error 1: `EPERM`: operation not permitted
- Error 22: `EINVAL`: invalid argument
- Error 517 does not exist but I guess the error code reports the error "ORed" with a high bit (512) that has a differenty meaning. Therefore 517-512 = 5 which is probably your  error code.
- Error 5: `EIO`: I/O error

HTH

AkazawaYun, 2025-01-07:
Thank you very much !!! You help me a lot ^ ^
Today I found out the Exception.Message is made in ExceptionHelper.cs of this dotnet/iot project, the 'errorCode' argument's value is actually from Marshal.GetLastWin32Error(), I realized it comes from linux but not c# _(:3/
It seems learning more about linux os can help me to solve these problem, and the links you given are very useful !

I think its too difficult to control these pin...orz rather than I throught... Maybe Raspberry Pi is easier according to the microsoft guide, but I buy a Niche Brand board because cheap... ( its seller suggest me to edit and compile the source code of the board  to config its gpio pins... cry)

raffaeler, 2025-01-07:
Great to hear that, thanks for the feedback.
The `GetLastWin32Error` is precious but it requires the PInvoke declaration to be specifically marked to report the error, otherwise it gets "lost".
Also, since the OS reports errors as numbers, it is always better to know how to decode them, rather than "hoping" that the error is marshalled back.

What are your difficulties now? Managing the GPIOs should not be hard at all!

Please remember that, although the pins looks all the same, you should always take a look at the manufacturer's documentation because not all the pins provides the same features (pull-up, notifications, etc.)

AkazawaYun, 2025-01-07:
Yes , I don't care how GetLastWin32Error works, as you said,  Knowning the meaning (decode) of error number is enough for this situation ^ ^

The biggest difficulty may come from the unique design of the board, it seems not to be a general linux os implentment ( I heard from its manufacturer ). I have two plan in my winter-holiday to solve it:

1. buy a digital-multimeter and some jumper wires to try an experiment;
2. buy a Standard linux os implentment divice , like Raspberry Pi, to find the different between them, I believe I can find out it As long as any one of the two board runs.

The documentation is saying some thing hard to do... I only learned c# yet..
https://github.com/tan-zhihao1/kickpi-book/blob/master/h618/en/31-Extended%20pin%20configuration.md
https://github.com/tan-zhihao1/kickpi-book/blob/master/h618/en/04-Hardware%20Factory%20Test.md

raffaeler, 2025-01-07:
Sounds like a great plan for the holiday!
The quality of the driver provided by the manufacturer is fundamental. For example, the `libgpiod` linux driver should be provided by the manufacturer to expose the hardware peripherals according to the `libgpiod` specifications. If there are problems there, end of the story. You should verify to have installed `libgpiod` for your board, otherwise you are using s`sysfs`.

Possibly do not rely on the `sysfs` driver (basically mapping the GPIOs on a file that you can read/write). We do support it, but the performance and the features are poor and less reliable.

For very simple tasks, or for tasks that requires real-time, you should consider using a microcontroller like a board based on ESP32. These devices are normally programmed with Arduino but you can use https://nanoframework.net/ to use .NET and C# there. But read carefully the limitations before. 
There are many possibilities, for example you can also split the tasks between ESP32 for real-time and hardware access on the microcontroller, and a linux board to "orchestrate" the devices.

It's up to your pleasure of programming, hacking and learning :-)

closed 2025-01-18 by dotnet-policy-service[bot]

krwq, 2025-04-24:
This is also duplicate of https://github.com/dotnet/iot/issues/671