dotnet/iot #2383: Ds18b20 temperature sensor

sterenas, 2025-02-19:
Hi,
Noobie question here. Couldn’t find the directory for the Ds18b20 temperature sensor. Does it mean dotnet doesn’t support it or is it possible to use another device (perhaps a generic one?) to get it to work?
Thanks for shedding some light on this.

raffaeler, 2025-02-19:
Welcome @sterenas .
The DS18B20 is a digital `1-wire` thermometer which can be directly interfaced to the `1-wire` bus on the Raspberry PI.
Our library does support the `1-wire` peripherals. See this doc:
https://github.com/dotnet/iot/blob/main/src/devices/OneWire/README.md
The linkedt `1-wire` sample shows how to use the DS18B20 sensor, but you should be able to drive any other sensor using the `1-wire` bus.

For details about the handshake, please refer to the official datasheet:
https://www.analog.com/media/en/technical-documentation/data-sheets/DS18B20.pdf

Enjoy

raffaeler, 2025-02-20:
[Triage] Please confirm that you could use the sensor as expected, otherwise let us know more details.

sterenas, 2025-02-20:
Thanks so much for your prompt reply.  I've set it all up but I'm getting the following error:

Found bus 'w1_bus_master1', scanning for devices ...
Access to the path '/sys/devices/w1_bus_master1/w1_master_search' is denied.

I'm sure I've ran the command:
sudo chmod a+rw /sys/bus/w1/devices/w1_bus_master1/w1_master_*

Still need to debug a bit more.

Thanks again for your help.  Will update as I make progress.
Sergio

sterenas, 2025-02-20:
...and here we go:

Temperatura en sonda 28-000000567e58: 26.69 °C
Found bus 'w1_bus_master1', scanning for devices ...
Found family 'Ds18b20' device '28-000000567e58' on 'w1_bus_master1'
Temperatura en sonda: 26.75 °C

Works perfect!

Thanks so much Raf!

closed 2025-02-20 by sterenas