dotnet/iot #2384: Error Toggling GPIO pin

rathbone01, 2025-02-24:
**Describe the bug**

When trying to use a GpioController and opening pin 10 as an output on a RasPi CM4, then call .Toggle I am getting an exception.

**Steps to reproduce**

```csharp
private async void BlinkLoop()
{
    if (controller is null)
    {
        _logger?.LogError("GpioController not initialized");
        return;
    }

    try
    {
        controller.OpenPin(actLedPin, PinMode.Output);

    }
    catch (Exception ex)
    {
        _logger?.LogError($"Error opening actLedPin: {ex.Message}");
    }

    try
    {
        while (!powerPanic)
        {
            controller.Toggle(actLedPin);
            await Task.Delay(500);
        }
    }
    catch (Exception ex)
    {
        _logger?.LogError($"Error in BlinkLoop: {ex.Message}: {ex.StackTrace}");
    }
}
```

**Expected behavior**

I expect that Gpio to toggle on and off.

**Actual behavior**

Get the following exception:
2025-02-21 08:35:49.307 -06:00 [ERR] Error in BlinkLoop: The given key '10' was not present in the dictionary.:    at System.Collections.Concurrent.ConcurrentDictionary`2.ThrowKeyNotFoundException(TKey key)
   at System.Collections.Concurrent.ConcurrentDictionary`2.get_Item(TKey key)
   at System.Device.Gpio.Drivers.Libgpiod.V1.LibGpiodV1Driver.Toggle(Int32 pinNumber)
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.Toggle(Int32 pinNumber)
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.Toggle(Int32 pinNumber)
   at System.Device.Gpio.GpioController.Toggle(Int32 pinNumber)
   at GapMonitor.Services.GpioService.BlinkLoop() in C:\Users\heatb\Desktop\GitHub_Repositories\DaTran-Fueling-Server\GapMonitor\Services\GpioService.cs:line 102

**Versions used**
8.0.8 - `dotnet --info` on the machine being used to build
8.0.8 - `dotnet --info` on the machine where app is being run (not applicable for self-contained apps)
3.2.0 - Version of `System.Device.Gpio` package
3.2.0 - Version of `Iot.Device.Bindings` package (not needed if bug is in `System.Device.Gpio`)

raffaeler, 2025-02-24:
Hi @rathbone01, sorry for the problem. 

Your exception shows `RaspberryPi3LinuxDriver` in the stack trace which means that you are using a RPi3 driver instead of a RPi4, which can most probably depends on the detection algorithm.

The Raspberry boards continuously change revision and model numbers and this may affect the ability to recognize the board. This in turn affects the driver detection algorithm. 

Please **provide the model number of your CM4** as described here:
http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/

The detection happens with the following model numbers: https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs#L155

Please post the **controller initialization** code too.

rathbone01, 2025-02-25:
Alright, I will provide that when I get a chance.
 I would like to add, in trying to fix some bugs before doing a deployment tomorrow I switched to .Write instead of .Toggle and that is functioning correctly.

Here is where I init:

```csharp       
        public bool Initalized { get; private set; } = false;

        private const int powerPanicPin = 14;
        private bool powerPanic = false;
        private int debounce = 10;

        private const int stResetPin = 27;
        private const int actLedPin = 10;
        private const int usbPowerControl = 3;
        private const int piPowerOverrride = 11;

        private ILogger<GpioService>? _logger;
        private GpioController? controller;
        private GapMonitorService? GapMonitorService;

        private List<int> relays = new() { 5, 6, 13, 19, 26, 7, 8, 9 }; // 1 is gpio 5 on rev D
        private IServiceProvider serviceProvider;

        public GpioService(ILogger<GpioService> logger, IServiceProvider serviceProvider)
        {
            _logger = logger;
            _logger.LogInformation("GpioService created");
            this.serviceProvider = serviceProvider;
        }

        public void Setup()
        {
            // if not on linux, don't setup the GPIO
            if (RuntimeInformation.IsOSPlatform(OSPlatform.Windows))
            {
                _logger?.LogInformation("Not on linux, skipping GPIO setup");
                return;
            }

            GapMonitorService = serviceProvider?.GetService<GapMonitorService>();
            if (GapMonitorService is null)
                _logger?.LogError("GpioService: LevelMonitorService is null");

            controller = new GpioController();

            // Setup power signal pin
            controller.OpenPin(usbPowerControl, PinMode.Output, PinValue.High);

            // Setup power override pin
            controller.OpenPin(piPowerOverrride, PinMode.Output, PinValue.Low);

            // Setup power panic pin
            controller.OpenPin(powerPanicPin, PinMode.InputPullUp);
            controller.RegisterCallbackForPinValueChangedEvent(powerPanicPin, PinEventTypes.Falling, PowerPanicPinChanged);

            foreach (int i in relays)
            {
                controller.OpenPin(i, PinMode.Output, PinValue.Low);
            }

            _logger?.LogInformation("GpioService setup");
            Initalized = true;

            // Start blinking light
            BlinkLoop();
        }
```

raffaeler, 2025-02-25:
The `Toggle` function uses `Write` to toggle the value.
https://github.com/dotnet/iot/blob/29639448a0a0bbf6a1841dea7b2fc76bf4dfa300/src/System.Device.Gpio/System/Device/Gpio/Drivers/Libgpiod/V1/LibGpiodV1Driver.cs#L244C5-L244C103

So the problem is that you probably does not initialize the pin by reading or writing an initial state.
After initializing the controller, set an initial value (or just read it) so that the `ConcurrentDictionary` contains the value for that pin. After this `Toggle` should work as expected.

rathbone01, 2025-02-25:
would this also work, setting it when it is opened?

controller.OpenPin(actLedPin, PinMode.Output, PinValue.High);

raffaeler, 2025-02-25:
Yes:
https://github.com/dotnet/iot/blob/29639448a0a0bbf6a1841dea7b2fc76bf4dfa300/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L175

rathbone01, 2025-02-25:
Alright thanks!

raffaeler, 2025-02-25:
@rathbone01 We would appreciate if you could run the diagnostics on your Raspberry to verify the id number of your CM4 board.
Thanks

rathbone01, 2025-02-25:
Probably more than you wanted, but here is the output of each of those commands:

[image]

FlowTran@CBMD-9d56:/ $ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Revision        : b03141
Serial          : 10000000dc9e3749
Model           : Raspberry Pi Compute Module 4 Rev 1.1

`FlowTran@CBMD-9d56:/ $ cat /proc/device-tree/model
Raspberry Pi Compute Module 4 Rev 1.1`

raffaeler, 2025-02-25:
Thank you. Apparently the "b03141" should match `0x3141` on our code but it doesn't.

@pgrawehr I believe the "b" pre-pending the revision of this CM4 board is killing the detection. What do you think?
https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs#L120

rathbone01, 2025-02-25:
Correct, the b will be included in the tryparse as a hex value

rathbone01, 2025-02-25:
```csharp
int.TryParse("b03141", NumberStyles.HexNumber, CultureInfo.InvariantCulture, out int firmware);
Console.WriteLine("0x" + firmware.ToString("X"));
string Connections = File.ReadAllText("connections.txt");
if (firmware == 0x3141)
{
    Console.WriteLine("Firmware is 0x3141");
}
else if (firmware == 0xB03141)
{
    Console.WriteLine("Firmware is 0xB03141");
}
else
{
    Console.WriteLine("Firmware is unknown");
}
```

gives the following output:
0xB03141
Firmware is 0xB03141

krwq, 2025-02-26:
@raffaeler

krwq, 2025-02-26:
btw there is masking in that logic

krwq, 2025-02-26:
Ok, I just did some quick checks to make sure this works as expected:
```
bool success = int.TryParse("b03141", NumberStyles.HexNumber, CultureInfo.InvariantCulture, out int firmware);
Console.WriteLine($"TryParse: {success}");
Console.WriteLine($"Firmware: {firmware:X2}");

var maskedFirmware = firmware & 0xFFFF;
Console.WriteLine($"Firmware (masked): {maskedFirmware:X2}");
```

and that worked as expected.

I believe the problem will be in the settings loading:
https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs#L230
I suspect either exception or suffix is an issue.

raffaeler, 2025-02-26:
> [@raffaeler](https://github.com/raffaeler) we should fix issue detection or create a new issue detailing what needs to be done before we close this

Totally agree, this is why I asked the OP to report the board result here.

Given the growing revision numbers, I believe it would be better to entirely change the detection algorithm and search for the model string instead:

> `cat /proc/device-tree/model Raspberry Pi Compute Module 4 Rev 1.1`

raffaeler, 2025-02-26:
@rathbone01 could you please, cut-paste this class and run the detection code method on your CM4 to understand if this fails?
https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/Gpio/RaspberryBoardInfo.cs#L230

Thank you

krwq, 2025-02-26:
Also if you find a simple way to fix please send it here (or send a PR)

pgrawehr, 2025-02-27:
@rathbone01 Can you provide the output of `gpioController.QueryComponentInformation()`? That will give us a hint what hardware is actually detected. The ´Raspberrry3Driver´ is definitelly not the problem, because that is used for the RPI4 as well, and as long as you're not using the built-in pull-up or pull-down feature of the GPIO lines, even a wrong detection will not have an effect. 

Also, I think we can remove the ´RaspberryBoardInfo´ or move it to a better place. At the moment, it's only use is to distinguish between "standard" and "CM" type boards, for which there should be an easier way. The information from this class might be interesting, but since it's fully internal at the moment, it's overdesigned.

rathbone01, 2025-03-04:
I didn't have the RaspberryBoardInfo object so I just returned and printed the dictionary of objects and got this output so its at least getting to the constructor:

processor.0: 0
BogoMIPS.0: 108.00
Features.0: fp asimd evtstrm crc32 cpuid
CPU implementer.0: 0x41
CPU architecture.0: 8
CPU variant.0: 0x0
CPU part.0: 0xd08
CPU revision.0: 3
processor.1: 1
BogoMIPS.1: 108.00
Features.1: fp asimd evtstrm crc32 cpuid
CPU implementer.1: 0x41
CPU architecture.1: 8
CPU variant.1: 0x0
CPU part.1: 0xd08
CPU revision.1: 3
processor.2: 2
BogoMIPS.2: 108.00
Features.2: fp asimd evtstrm crc32 cpuid
CPU implementer.2: 0x41
CPU architecture.2: 8
CPU variant.2: 0x0
CPU part.2: 0xd08
CPU revision.2: 3
processor.3: 3
BogoMIPS.3: 108.00
Features.3: fp asimd evtstrm crc32 cpuid
CPU implementer.3: 0x41
CPU architecture.3: 8
CPU variant.3: 0x0
CPU part.3: 0xd08
CPU revision.3: 3
Revision: b03141
Serial: 10000000dc9e3749
Model: Raspberry Pi Compute Module 4 Rev 1.1``

pgrawehr, 2025-03-06:
@rathbone01 That's not what I meant. 

Please run `var s=controller.QueryComponentInformation().ToString()` and post the output of that. That shows the actual driver that was constructed.

rathbone01, 2025-03-14:
``sudo dotnet PiGpioTestApp.dll
{System.Device.Gpio.GpioController} Generic GPIO Controller
 └{System.Device.Gpio.Drivers.RaspberryPi3Driver} Generic Raspberry Pi Wrapper driver
  └{System.Device.Gpio.Drivers.RaspberryPiCm3Driver} Raspberry Pi Compute Module 4 Rev 1.1 linux driver with 48 pins and an interrupt driver
   └{System.Device.Gpio.Drivers.Libgpiod.V1.LibGpiodV1Driver} LibGpiodDriver

pgrawehr, 2025-03-18:
Ok, that looks as expected. Thus there seems to be no other issue than the crash with the toggle call. I'll  fix that.

andreasgrosche, 2025-03-31:
I get the same exception on Raspberry Pi Zero 2 W:
```
Unhandled exception: System.Collections.Generic.KeyNotFoundException: The given key '4' was not present in the dictionary.
   at System.Collections.Concurrent.ConcurrentDictionary`2.ThrowKeyNotFoundException(TKey key)
   at System.Collections.Concurrent.ConcurrentDictionary`2.get_Item(TKey key)
   at System.Device.Gpio.Drivers.Libgpiod.V1.LibGpiodV1Driver.Toggle(Int32 pinNumber)
   at System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver.Toggle(Int32 pinNumber)
   at System.Device.Gpio.Drivers.RaspberryPi3Driver.Toggle(Int32 pinNumber)
   at System.Device.Gpio.GpioPin.Toggle()
```

I'm looking forward to get this fixed with your commit
> Ok, that looks as expected. Thus there seems to be no other issue than the crash with the toggle call. I'll fix that.

Is there an estimation when this fix will be released?

pinout information:

[image]

`controller.QueryComponentInformation().ToString()` prints:
```
{System.Device.Gpio.GpioController} Generic GPIO Controller
 └{System.Device.Gpio.Drivers.RaspberryPi3Driver} Generic Raspberry Pi Wrapper driver
  └{System.Device.Gpio.Drivers.RaspberryPi3LinuxDriver} Raspberry Pi Zero 2 W Rev 1.0 linux driver with 28 pins and an interrupt driver
   └{System.Device.Gpio.Drivers.Libgpiod.V1.LibGpiodV1Driver} LibGpiodDriver
```

pgrawehr, 2025-03-31:
@andreasgrosche We're planning a release towards the end of this week, if nothing unexpected happens. For the problem above, there's also an easy workaround: If you set the initial value once, the Toggle call will work.

closed 2025-04-03 by joperezr