dotnet/iot #2393: Add single bit writing function to SpiDevice

liziyu0714, 2025-04-28:
SpiDevice.Write or SpiDevice.WriteByte just write a full byte to device, but this time I need to write a **Single** bit to device. (I am using a 3-lane mode for ST77916, which write mode began with a single bit that represents D/CX to distinguish command and data)

Is this a function to be implemented in the future?

raffaeler, 2025-04-28:
Hi @liziyu0714 I don't believe the SPI hardware can emit a single bit.
According to the [SPI Raspberry PI documentation](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#supported-bits-per-word):
```
Supported bits per word
8 - normal
9 - supported using LoSSI mode
```

Said that, I can think to three different workarounds here:
1. Set the DataBitLength = 9 and send the 9th bit to one. This may do what you are looking for. The `write` overload that takes a buffer should do that. You may clone the SPI code from this repo and trying to modify it, if you believe that it does not work correctly with the 9 bits length settings. In this case, please give us feedback!

2. Send a 0x80 or 0x01 (depending whether you want the first bit at the top or the bottom of the serial communication). Since the other bits are zeros, your device hardware **may** (or not) accept this as it was a single bit.

3. Disable the SPI, manually send the single bit using the GPIO and immediately after re-enable the SPI to continue the communication. Depending on the time taken by the Raspberry PI to enable the SPI, it may work correctly.

I strongly suggest to observe the result using an oscilloscope (if you have one :-) ), so that you can observe the correctness of the sequence for the three cases.

liziyu0714, 2025-05-01:
I tried 

> Set the DataBitLength = 9 and send the 9th bit to one. This may do what you are looking for. The write overload that takes a buffer should do that. You may clone the SPI code from this repo and [...]

But the runtime had thrown an exception:
> Unhandled exception. System.IO.IOException: Error 22. Can not set SPI data bit length to 9.
   at System.Device.Spi.UnixSpiDevice.Initialize()
   at System.Device.Spi.UnixSpiDevice.Write(ReadOnlySpan`1 buffer)
   at ST77916.Write(SpiDevice spi, ST77916Command command) in /---/ST77916.cs:line 44
   at Program.<Main>$(String[] args) in /---/Program.cs:line 76

**When I set the DataBitLength to 8, and the exception disappeared.**

this is my code : 
Program.cs
```csharp
using System.Device;
using System.Device.Spi;

SpiConnectionSettings spiConnectionSettings = new SpiConnectionSettings(0, 0){
    ClockFrequency = 40000000,
    Mode = SpiMode.Mode0,
    DataBitLength = 9,
};
Console.WriteLine("Creating SpiDevice...");
SpiDevice st77916 = SpiDevice.Create(spiConnectionSettings);
Console.WriteLine("Creating SpiDevice done.");

Console.WriteLine("Start to init...");

ST77916Command[] initcommands = [
.......
];

foreach (var command in initcommands)
{
    ST77916.Write(st77916,command);
}
```

ST77916.cs
``` csharp
using System;
using System.Buffers;
using System.Device.Gpio;
using System.Device.Spi;

public struct ST77916Command
{
    public byte Command;
    public byte[] Data;
    public ST77916Command(byte command, byte[] data)
    {
        Command = command;
        Data = data;
    }
};

public  class ST77916
{
    public static void Write(SpiDevice spi, ST77916Command command)
    {
        byte[] buffer = new byte[2];
        buffer[0] = 0x00;
        buffer[0] |= (byte)(command.Command >> 1);
        buffer[1] |= (byte)(command.Command << 7);
        spi.Write(buffer);
        foreach(byte b in command.Data)
        {
            buffer[0] = 0x80;
            buffer[0] |= (byte)(b >> 1);
            buffer[1] |= (byte)(b << 7);
            spi.Write(buffer);
        }
    }
}
```

liziyu0714, 2025-05-01:
My device is Raspberry Pi 4B, with the newest 64bit raspbian.

raffaeler, 2025-05-01:
@liziyu0714 
The error comes directly from the native primitive (`ioctl` C function) of the Raspberry PI, not from the .NET code.

As I wrote above, the 9-bit configuration is `supported using LoSSI mode`. 
The error `22` means "Invalid Parameter", therefore there is some other settings that is not compatible with the `9-bit` mode.

You can start trying by changing `Mode` (start with `Mode3`) and the lowering the speed. 

HTH

liziyu0714, 2025-05-02:
Should I change my raspberry pi's config to open LoSSI mode?

raffaeler, 2025-05-02:
From what I read, you just need to enable the SPI (as you already did) using `raspi-config`. 
Beyond that, the `Mode3` should be enough, but I never tried this.

Basically the Ioctl used in our code just set the register of the hardware peripherals, bypassing Linux.

The registers for the SPI peripheral, including LoSSI mode, are described here:
- https://github.com/raspberrypi/linux/issues/3348
- https://datasheets.raspberrypi.com/bcm2835/bcm2835-peripherals.pdf starting from page 152, paragraph 10.5

krwq, 2025-05-08:
[Triage] Can you try [SoftwareSpi](https://github.com/dotnet/iot/blob/main/src/devices/SoftwareSpi/SoftwareSpi.cs) and see if you can unblock yourself? Regardless we should still make sure we don't have a bug on our side

raffaeler, 2025-05-08:
@liziyu0714 
I see now `40000000` speed, which is way too fast (LoSSI mode is low speed btw).
You should be able to change speed right after if you need it.

krwq, 2025-05-22:
[Triage] any updates @liziyu0714?

closed 2025-07-08 by dotnet-policy-service[bot]