dotnet/iot #2396: Access to GPIO fails with all versions of RaspiOS starting from 2025-05-06 on Raspberry Pi 5

GreenShadeZhang, 2025-05-12:
**Describe the bug**

I'm having issues with GPIO pin initialization failing on the latest version of the Raspberry Pi, the Raspberry Pi 5. I've seen many people have the same issue and I'd like to know if the latest 4.0 version of the library has fixed this issue.

```csharp
Unhandled exception. System.IO.IOException: Error setting pin mode, for pin: 18, error: 16
   at System.Device.Gpio.Drivers.Libgpiod.V1.LibGpiodV1Driver.SetPinMode(Int32 pinNumber, PinMode mode)
   at System.Device.Gpio.Drivers.LibGpiodDriver.SetPinMode(Int32 pinNumber, PinMode mode)
   at System.Device.Gpio.GpioController.SetPinMode(Int32 pinNumber, PinMode mode)
   at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber, PinMode mode)
   at Verdure.Iot.Device.LcdConfig..ctor(SpiDevice spi, SoftwarePwmChannel pwmBacklight, Int32 spiFreq, Int32 rst, Int32 dc, Int32 bl, Int32 blFreq) in C:\Users\gil\Music\github\dotnet-iot-tutorial-code\src\Verdure.Iot.Device\Lcds\LcdConfig.cs:line 28
   at Verdure.Iot.Device.LCD2inch4..ctor(SpiDevice spi, SoftwarePwmChannel pwmBacklight, Int32 spiFreq, Int32 rst, Int32 dc, Int32 bl, Int32 blFreq) in C:\Users\gil\Music\github\dotnet-iot-tutorial-code\src\Verdure.Iot.Device\Lcds\LCD2inch4.cs:line 11
   at Program.<Main>$(String[] args) in C:\Users\gil\Music\github\dotnet-iot-tutorial-code\src\Samples\Iot.Lcd\Program.cs:line 27
Aborted
```

```csharp
gil@pi5:~/Lcd24 $ gpiodetect
gpiochip0 [pinctrl-rp1] (54 lines)
gpiochip10 [gpio-brcmstb@107d508500] (32 lines)
gpiochip11 [gpio-brcmstb@107d517c00] (15 lines)
gpiochip12 [gpio-brcmstb@107d517c20] (6 lines)
gpiochip13 [gpio-brcmstb@107d508520] (4 lines)
gpiochip0 [pinctrl-rp1] (54 lines)
```

```csharp
gil@pi5:~/Lcd24 $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```csharp
gil@pi5:~/Lcd24 $ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

Revision        : b04170
Serial          : 1de2fc6e847aef57
Model           : Raspberry Pi 5 Model B Rev 1.0
```

**Steps to reproduce**

```csharp
// put sample code here
```

**Expected behavior**

**Actual behavior**

**Versions used**
iot version 3.2.0

```csharp
gil@pi5:~/Lcd24 $ dotnet --info
.NET SDK:
 Version:           8.0.408
 Commit:            e6663e9b3d
 Workload version:  8.0.400-manifests.22ac4dbc
 MSBuild version:   17.11.26+2b19be476

Runtime Environment:
 OS Name:     debian
 OS Version:  12
 OS Platform: Linux
 RID:         linux-arm64
 Base Path:   /home/gil/.dotnet/sdk/8.0.408/

.NET workloads installed:
Configured to use loose manifests when installing new manifests.
There are no installed workloads to display.

Host:
  Version:      8.0.15
  Architecture: arm64
  Commit:       50c4cb9fc3

.NET SDKs installed:
  8.0.408 [/home/gil/.dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 8.0.15 [/home/gil/.dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 8.0.15 [/home/gil/.dotnet/shared/Microsoft.NETCore.App]

Other architectures found:
  None

Environment variables:
  DOTNET_ROOT       [/home/gil/.dotnet]

global.json file:
  Not found

Learn more:
  https://aka.ms/dotnet/info

Download .NET:
  https://aka.ms/dotnet/download
```

raffaeler, 2025-05-12:
Hi @GreenShadeZhang 
from previous issues we learned that certain boards (the earliest) of the RPi5 used the `gpioChip 4` and other models `gpioChip 0`

Could you please try specifying the chip number manually, using `0` as for your diagnostics?

```csharp
var gpioController = new GpioController(new LibgpiodDriver(0)); 
```

GreenShadeZhang, 2025-05-12:
> Hi [@GreenShadeZhang](https://github.com/GreenShadeZhang) from previous issues we learned that certain boards (the earliest) of the RPi5 used the `gpioChip 4` and other models `gpioChip 0`
> 
> Could you please try specifying the chip number manually, using `0` as for your diagnostics?
> 
> var gpioController = new GpioController(new LibgpiodDriver(0));

I think I found the real reason. I used my code to test the 2024 0704 version of the system, and this problem also occurred. I also tried to use the specified chip number as you said, and the problem also occurred.

But when I was doing unit testing, when I only created one GPIO controller object, this problem would not occur regardless of the new or old system.

Therefore, I adjusted my code, that is, only define a global GPIO controller object, so that it is passed in wherever it is used, and this problem will not occur.

The effect of using Raspberry Pi 5 is as follows:

[image]

My main code is as follows:

https://github.com/GreenShadeZhang/dotnet-iot-tutorial-code/blob/main/src/Samples/Iot.Lcd/Program.cs

```csharp
using SixLabors.ImageSharp;
using SixLabors.ImageSharp.PixelFormats;
using SixLabors.ImageSharp.Processing;
using System.Device.Gpio;
using System.Device.Pwm.Drivers;
using System.Device.Spi;
using Verdure.Iot.Device;

var gpio = new GpioController();

using var pwmBacklight = new SoftwarePwmChannel(pinNumber: 18, frequency: 1000, controller: gpio);

pwmBacklight.Start();

string input2inch4Path = "LCD_2inch4.jpg";

string input1inch47Path = "verdure901.png";

using SpiDevice sender2inch4Device = SpiDevice.Create(new SpiConnectionSettings(0, 0)
{
    ClockFrequency = 50000000,
    Mode = SpiMode.Mode0
});
using SpiDevice sender1inch47Device = SpiDevice.Create(new SpiConnectionSettings(0, 1)
{
    ClockFrequency = 50000000,
    Mode = SpiMode.Mode0
});

using var inch24 = new LCD2inch4(sender2inch4Device, pwmBacklight, gpio);
inch24.Reset();
inch24.Init();
inch24.SetWindows(0, 0, LCD2inch4.Width, LCD2inch4.Height);
inch24.Clear();
inch24.BlDutyCycle(50);

using var inch147 = new LCD1inch47(sender1inch47Device, pwmBacklight, gpio);

//inch147.Reset();
inch147.Init();
inch147.SetWindows(0, 0, LCD1inch47.Width, LCD1inch47.Height);
inch147.Clear();
inch147.BlDutyCycle(50);

byte[] data1 = [];

byte[] data2 = [];

using (Image<Bgra32> image2inch4 = Image.Load<Bgra32>("LCD_2inch.jpg"))
{
    image2inch4.Mutate(x => x.Rotate(90));
    using Image<Bgr24> converted2inch4Image = image2inch4.CloneAs<Bgr24>();
    data1 = inch24.GetImageBytes(converted2inch4Image);
}
//await Task.Delay(50);
Console.WriteLine("2inch4 Done");

using (Image<Bgra32> image1inch47 = Image.Load<Bgra32>(input1inch47Path))
{
    using Image<Bgr24> converted1inch47Image = image1inch47.CloneAs<Bgr24>();
    //inch147.ShowImageData(converted1inch47Image);
    data2 = inch147.GetImageBytes(converted1inch47Image);
}
//await Task.Delay(50);
Console.WriteLine("1inch47 Done");

while (true)
{
    inch24.ShowImageBytes(data1);

    inch147.ShowImageBytes(data2);
    //using (Image<Bgra32> image2inch41 = Image.Load<Bgra32>(input2inch4Path))
    //{
    //    using Image<Bgr24> converted2inch4Image1 = image2inch41.CloneAs<Bgr24>();
    //    inch24.ShowImageData(converted2inch4Image1);
    //}

    //Console.WriteLine("2inch41 Done");

    //using (Image<Bgra32> image1inch471 = Image.Load<Bgra32>("verdure901.png"))
    //{
    //    using Image<Bgr24> converted1inch47Image1 = image1inch471.CloneAs<Bgr24>();
    //    inch147.ShowImageData(converted1inch47Image1);
    //}

    //Console.WriteLine("1inch471 Done");
}
//Console.ReadLine();
```

raffaeler, 2025-05-12:
Sounds great @GreenShadeZhang 
Please refer to [this document](https://github.com/dotnet/iot/blob/204a5bd02658071fa33fd24cf35d7d7659269d13/Documentation/Devices-conventions.md#lifetime-management-or-what-should-be-disposed-by-what) for the correct usage of multiple `GpioController` and when/how dispose the objects controlling the devices.

You may find useful [this other document](https://github.com/dotnet/iot/blob/204a5bd02658071fa33fd24cf35d7d7659269d13/src/devices/Gpio/README.md#usage-of-virtual-gpiocontroller-and-virtual-gpiopin) as well.

HTH

GreenShadeZhang, 2025-05-13:
Thank you for the documents you provided. I need to spend some time testing the code to make it work well on both the Raspberry Pi 5 and the Raspberry Pi Zero 2W. I also hope that my situation can provide some reference for others.

GreenShadeZhang, 2025-05-15:
@raffaeler I am asking an unrelated question here.

When will the 4.0 version of the nuget package be uploaded to the official nuget website? Currently, I find that the latest version is still 3.2.0.

pgrawehr, 2025-05-20:
V4.0.1 has now finally been shipped. Sorry for the delay, we were experiencing issues with signing certificates.

krwq, 2025-06-05:
[Triage] @GreenShadeZhang were you able to get this resolved? Can this issue be closed?

GreenShadeZhang, 2025-06-05:
Ok, I will close this question.