dotnet/iot #2404: Interrupt handler for Mcp23xxx only handles interrupts on first 4 pins

HakanL, 2025-07-13:
**Describe the bug**

In the Mcp23xxx driver (used by Mcp23008, Mcp23017, etc) the interrupt handler has a loop where it checks if the mask is matching the pending interrupts, but the loop checks only 4 of the total 8 pins, so interrupts are never working on pins 5-8 (for either port A and B on Mcp23017). This is the code:
```
            int mask = 1;
            int pin = 0;

            while (mask < 0x10)
            {
                if ((interruptPending & mask) != 0)
                {
                    CallHandlerOnPin(pin + offset, newValues & mask);
                }

                mask = mask << 1;
                pin++;
            }
```

**Steps to reproduce**

Hook up a button to pin 5-8 and you'll never receive the interrupts.

**Expected behavior**

Interrupts should received

**Actual behavior**

Nothing

**Versions used**

4.0.1

HakanL, 2025-08-14:
@raffaeler Do you have any insight into when this fix (that has been merged) will make it into a pre-release package on Nuget?

raffaeler, 2025-08-14:
Hi @HakanL Not sure, let me ask to @joperezr about the pre release packages ...

joperezr, 2025-08-28:
AFAIK there is nothing really blocking us in this. @raffaeler lets discuss this tomorrow in our sync

HakanL, 2025-09-03:
That worked great, thanks!