dotnet/iot #2415: Testing construction for null in libgpioddriver

felixcollins, 2025-08-27:
**Describe the bug**

https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/Gpio/Drivers/LibGpiodDriver.cs#L232

OpenPin tests the construction of the LineHandle for null.  This test can never fail. Unfortunately, this then hides the problem with opening the pin using the libgpiod behind an incorrectly thrown ObjectDisposed exception later on. 

**Steps to reproduce**
By inspection the code is wrong - a constructor can never return null (Only Jon Skeet can make it do that)

int pin = <choose a pin number that causes the underlying LibgpiodV1.gpiod_chip_get_line call to fail>
var drv = new LibGpiodDriver(0); // gpiochip0
using var gpioController = new GpioController(drv);
gpioController.OpenPin(pin)

**Expected behavior**

It should throw ArgumentException or other relevant exception after reading and decoding LastError on the failed call. General guidance - any calls to interop methods should check for failure at the point of the call.

**Actual behavior**

In my case it threw a mysterious ObjectDisposed exception later on at 
at System.Device.Gpio.Libgpiod.V1.LineHandle.get_Handle()

Throwing ObjectDisposed because the underlying native handle is null is also incorrect and misleading.

**Versions used**

felixcollins, 2025-08-29:
@krwq Where do you have general conversations about this library? I opened the System.Device.Gpio  solution and found that it takes a dependency on the Iot.Device.Bindings project. There are tests for MCP3008. Why is that in there?

raffaeler, 2025-08-29:
@felixcollins we have a [discord server](https://discord.gg/kqWhBbj) for discussions.

>  I opened the System.Device.Gpio solution and found that it takes a dependency on the Iot.Device.Bindings project. There are tests for MCP3008. Why is that in there?

Anyway, could you please elaborate this?
The `System.Device.Gpio` projects does not take any dependency on other libraries from this repository. It contains the "building blocks" that are needed from all the other projects that are shipped in the `Iot.Device.Bindings`.
The `MCP3008` binding is inside the `Mcp3xxx` project which depends on the `System.Device.Gpio`.
The `Iot.Device.Bindings` is just a project that conveniently aggregate all of the bindings projects that can be developed and tested separately.

PR #2418 (2025-09-04): Fixes #2415 - Testing construction for null in libgpioddriver
fixes #2415

Now checking for IntPtr.Zero which was probably the original intention.
 ###### Microsoft Reviewers: [Open in CodeFlow](https://microsoft.github.io/open-pr/?codeflow=https://github.com/dotnet/iot/pull/2418)
commit ae707ec8 (2025-09-04): Fixes #2415 - Testing construction for null in libgpioddriver (#2418)

* fixes #2415 - Checking for a zero returned from interop instead of a null returned from constructor (which is not possible).

* Removing public setter of Handle to make LineHandle slightly safer.

* Moved null pointer check to LineHandle ctor. Removed unused IsInvalid method. Call ReleaseLock from Dispose to remove duplicated code.
closed 2025-09-04 by krwq