dotnet/iot #2419: 4 times Thread.Sleep(1) in Nrf24l01 Send ... why?

jblahos, 2025-09-01:
Hello,

I was looking at this library (Iot.Device.Bindings), considering to use it with my raspberry pi project. I want to use the Nrf24l01 wrapper. But when I checked it out internally, I noticed that its Send() method has 4x Thread.Sleep(1) inside it. This would massively slow down my sending rate. I want my nrf to only send data, never receive anything, and I want to send data rapidly. So why is the Send implemented like this, switching back to receive after every send and having those sleeps?

[            Thread.Sleep(1);](https://github.com/dotnet/iot/blob/945495e9fb4c48e964b83d87127befd31fddfb25/src/devices/Nrf24l01/Nrf24l01.cs#L184)

I would suggest that in Send there should be simple write of the data and nothing else.

pgrawehr, 2026-10-15:
@jblahos [Triage] There's a draft PR open with the requested changes. We currently have no way of testing it, so would it be possible for you to test the new wait times? Please let us know if you need help in running the PR code.

jblahos, 2026-10-16:
Hi, sorry, I cannot test this anymore. I moved away from this lib to a different solution (c++ rf24 lib imported into C# app through a thin dll wrapper).